collin80/SavvyCAN #488: Cannot find ID in DBC file(s). Not adding graph.

catphish, 2022-08-26:
Since upgrading to Linux Mint 21 (Ubuntu 22.04) I am unable to create graphs. I have DBC files configured and the data is interpreted correctly, but when attempting to create a graph, I get the error "Cannot find ID [ID] in DBC file(s). Not adding graph."

collin80, 2022-08-31:
Ah... so it does... Here's the thing: in theory I wrote the code that does this. I don't remember doing it. I had no idea you could do that with the program... I guess I ought to go fix it and maybe find out if it was ever even completed. The usual way to graph things is to bring up the window manually, right click the graphing window, and say "Add New Graph" but the way you're trying to use would be really handy if it worked.

catphish, 2022-08-31:
Sorry, I don't think I was sufficiently clear about the situation. I was using this functionality previously, right-clicking on a specific signal and clicking "add to ... graphing window".

It's not intuitive at all, but it worked and was very convenient. However since upgrading my OS, the feature has stopped working. I get the error I mentioned instead.

Personally I find this behaviour very useful, much better than if it graphed every signal in the message, as I usually want to compare signals from different messages.

collin80, 2022-09-01:
I tried this out some more and it works for me pretty reliably. However, my guess is that you might have a higher DPI monitor. I think between Ubuntu and the QT people there were changes to how things work on hi-dpi monitors. Currently the way it figures out which signal you are hovering over is silly. I'm surprised it works consistently for me. The real solution here is probably to not just expand the height of the row while having multiple lines of text in it. It would probably be better to have multiple rows for a message if it has signals in it. However, that doesn't play so well with the way QT works. So, I settled on the current "creative" solution but I don't like it. I may investigate perhaps using a "tree" style view  in the "Data" column. This would allow me to more easily figure out what exactly you're over top of when right clicking. Anyway, that's a long way of saying "it works sometimes but it's way too brittle and way too unintuitive". It needs a lot of work.