collin80/SavvyCAN #489: Connetced devices do not persist

catphish, 2022-08-26:
Using SavvyCAN on Linux I add a QT SerialBus Device (socketcan, can0), however this setting isn't persisted. Every time I open the application I have to re-add the device. Clicking "Save Bus Settings" doesn't seem to have any impact. This isn't a huge problem but it would save a lot of time if this configuration could be saved.

catphish, 2022-08-31:
Thank you, it would be very convenient if this worked :slightly_smiling_face:

jsphuebner, 2023-09-05:
Good to see it's not just me ;) Just compiled the latest code and it is not persistent. I think it is saved to .config/EVTV/SavvyCAN.ini under the [connections] section but then not loaded

collin80, 2023-09-06:
Ooops... well, the above fixes never happened. I guess it slipped my mind as I nearly never want to retain the list of previously connected adapters and so I haven't been annoyed by this enough to fix it. I've labeled it as a bug now, hopefully that will help to remind me to work on this. Sadly I just haven't worked on SavvyCAN that much in the last couple of years. I'm hoping to get back to it some time soon.