collin80/SavvyCAN #493: Custom frame sender, adding 0x00 to end of data field

wakey1991, 2022-09-04:
After making an 8 byte data field in the correct format, most of the time (although not every time) when you press enable, it will add usually one extra set of 0x00 at the end of the data field. If you don't spot it, the message is disregarded on the bus due to its length.

Its manageable but annoying, is it something i am doing wrong or just a bug?

I haven't found a specific pattern other than after making a change in the data field.

Thanks

AndyHuska, 2022-09-08:
I found the same issue.  In some cases there is a parser that runs to reformat the data and if there is a trailing space it turns it into an extra 0x00.  I will attempt to commit a fix and make a pull request....

AndyHuska, 2022-09-17:
The issue is on line 901 of framesenderwindow.cpp, change it to ignore empty tokens in the string parsing and the trailing 0x00 will go away when editing data or loading saved messages:
tokens = ui->tableSender->item(line, 6)->text().split(" ", QString::SkipEmptyParts);

Attempting to make a pull request now...

AndyHuska, 2022-09-17:
PR submitted: https://github.com/collin80/SavvyCAN/pull/494