collin80/SavvyCAN #497: support for panda protocol

freeesty, 2022-09-18:
It would be nice to read live can data from CANserver in SavvyCAN.
Canserver supports raw data upload with panda protocol.

more information here:
https://github.com/joshwardell/CANserver/wiki/PandaProtocol

collin80, 2022-09-20:
The format doesn't seem too difficult to support. At the moment my copy of the source is all torn up in the connection code. I'm trying to expand it and support some other interfaces. But, then I got busy. So, I can't quickly add this support but I would like to do so when I get the connection code smoothed out again and ready for more additions. As a bonus, this seems like it would extend support for the Comma.AI panda as well. Being able to support more such devices is always a good thing.

pyjamasam, 2022-09-21:
I have some code that does this from my older fork.  I'll see about pulling an up to date branch and trying to merge it in against that.