collin80/SavvyCAN #531: CAN speed 83.333 kb / s

Magneelektronics, 2022-10-09:
Hello to all colleagues, I started working on savyycan, this is very good software :) but I do not see the choice of CAN speed 83.333 kb / s used often in Mercedes cars, now I am trying to add this speed to the savyycan panel but I have a problem to work with savyycan on windows 10 visual studio2022, can you help me with this problem? thank you in advance for your help

collin80, 2022-10-11:
It should be possible to set this speed now as of 005a2aa56c6486808a3db9f0099682e4c97540aa so the continuous build will soon have this commit compiled in and ready to test.

Magneelektronics, 2022-10-12:
Super Collin, I don't know why but I have a problem with github in google hrom, now I logged in edge and I can add a message, I would try to add 83.33 speed to the code myself, but I can't cope with the savyycan compilation on windows 10 64 :(

Magneelektronics, 2022-10-12:
I also noticed a problem with sending CAN data, even when we finish sending, the frames are still sent, we have to disconnect the power from arduino due to end the data transmission, it's a big problem and have you tried to fix it?