collin80/SavvyCAN #539: Playback CLX000 log files not working properly with "Use original frame timing" option set.

5abina, 2022-11-05:
Hello,

like in topic playback of CLX000 log files not working with "Use original frame timing" option set.

Playback of CLX000 seems to work properly when "Use original frame timing" option is not set.

Is is possible to resolve this? I'm using Windows 11 with latest updates.

Best regards
Daniel

5abina, 2022-11-10:
Hello, here are the beginning of the file:

# Logger type: 29
# HW rev: 00.01
# FW rev: 01.04.02
# Logger ID: 8485b4ef
# Session No.: 45
# Split No.: 1
# Time: 20221012T105555
# Value separator: " "
# Time format: 3
# Time separator: ""
# Time separator ms: ""
# Date separator: ""
# Time and date separator: "T"
# Bit-rate: 250000
# Silent mode: false
# Cyclic mode: true
Timestamp ID Data
105555932 cff0051 fffffffffffffff3
105555942 cff0052 fffffffffffffff3
105555952 cff0051 fffffffffffffff3
105555962 cff0052 fffffffffffffff3
105555972 cff0051 fffffffffffffff3
105555982 cff0052 fffffffffffffff3
105555987 14ff023d f000f0f0ffffffff
105555992 cff0051 fffffffffffffff3
105556002 cff0052 fffffffffffffff3
105556012 cff0051 fffffffffffffff3
105556022 cff0052 fffffffffffffff3
105556032 cff0051 fffffffffffffff3
105556042 cff0052 fffffffffffffff3
105556052 cff0051 fffffffffffffff3
105556062 cff0052 fffffffffffffff3
105556072 cff0051 fffffffffffffff3
105556082 cff0052 fffffffffffffff3
105556087 14ff023d f000f0f0ffffffff
105556092 cff0051 fffffffffffffff3
105556102 cff0052 fffffffffffffff3
105556112 cff0051 fffffffffffffff3
105556122 cff0052 fffffffffffffff3
105556132 cff0051 fffffffffffffff3
105556142 cff0052 fffffffffffffff3
105556152 cff0051 fffffffffffffff3
105556162 cff0052 fffffffffffffff3
105556172 cff0051 fffffffffffffff3
105556182 cff0052 fffffffffffffff3
105556187 14ff023d f000f0f0ffffffff
105556192 cff0051 fffffffffffffff3
105556202 cff0052 fffffffffffffff3
105556212 cff0051 fffffffffffffff3
105556222 cff0052 fffffffffffffff3
105556232 cff0051 fffffffffffffff3
105556242 cff0052 fffffffffffffff3
105556252 cff0051 fffffffffffffff3
105556262 cff0052 fffffffffffffff3
105556272 cff0051 fffffffffffffff3
105556282 cff0052 fffffffffffffff3
105556287 14ff023d f000f0f0ffffffff
105556292 cff0051 fffffffffffffff3
105556302 cff0052 fffffffffffffff3
105556312 cff0051 fffffffffffffff3
105556322 cff0052 fffffffffffffff3
105556332 cff0051 fffffffffffffff3
105556342 cff0052 fffffffffffffff3
105556352 cff0051 fffffffffffffff3
105556362 cff0052 fffffffffffffff3
105556363 18ff0152 0000ffffffffffff
105556372 cff0051 fffffffffffffff3
105556373 18ff0151 0000ffffffffffff
105556382 cff0052 fffffffffffffff3
105556383 18ec3d52 20100003ff5200ff
105556387 14ff023d f000f0f0ffffffff
105556392 cff0051 fffffffffffffff3
105556402 cff0052 fffffffffffffff3
105556412 cff0051 fffffffffffffff3
105556422 cff0052 fffffffffffffff3
105556432 cff0051 fffffffffffffff3
105556433 18ebff52 01c704700a00f703
105556433 18ec3d51 20100003ff5100ff
105556442 cff0052 fffffffffffffff3
105556452 cff0051 fffffffffffffff3
105556462 cff0052 fffffffffffffff3
105556472 cff0051 fffffffffffffff3
105556482 18ebff51 012a04700b00ec03
105556483 cff0052 fffffffffffffff3
105556483 18ebff52 026c1059a0c8c100
105556487 14ff023d f000f0f0ffffffff
105556492 cff0051 fffffffffffffff3
105556502 cff0052 fffffffffffffff3
105556512 cff0051 fffffffffffffff3
105556522 cff0052 fffffffffffffff3
105556532 cff0051 fffffffffffffff3
105556532 18ebff51 02671057a0c80910

collin80, 2023-01-18:
I just fixed some serious problems with the loading of CLX files. I think it is very likely that it solved this issue as well. In fact, I just tested "use original frame timing" mode with a CLX file and it worked well for me. I'm going to close this issue as it seems to be fixed as of 202756c667e8a7ae1f33dac8c476a074ab89774c

closed 2023-01-18 by collin80