collin80/SavvyCAN #541: Tactrix Openport 2.0 won't connect

FLcruising, 2022-11-10:
Replaced a dead laptop with Tactrix OP working with only v1.0.245. Later versions never found the device. New laptop shows com port 3 until I install the device driver's for the passthru J2534, then SavvyCAN has no com port listed. How do I get this working?

ojuniour, 2023-03-28:
get the latest version...if 64 bit isn't working try compiling for 32 bit. (I can also compile one for ya if you want)

FLcruising, 2023-03-28:
Thank you. I found a 32-bit version that does work. V207

closed 2023-03-28 by FLcruising

ojuniour, 2023-03-29:
> Thank you. I found a 32-bit version that does work. V207
> 
> [#327 (comment)](https://github.com/collin80/SavvyCAN/discussions/327#discussioncomment-4204040)

Alright.. Just so you know, there are few bugs and added features in the latest version also. 
@collin80 is it possible to also include 32bit version in any future updates ? 🙏Open port 2.0 device driver doesn't seem to work too well with 64 bit version (on windows).