collin80/SavvyCAN #542: IXXAT USB to CAN v2

pink-fr3ud, 2022-11-21:
Hi ,
I'm trying to make my IXXAT interface work with SavvyCAN on a Windows system. I know this interface is compatible socketcan on Linux, but on windows it uses VCI4.
Someone already did a QtSerialBus plugin working on windows for this : https://github.com/gmarescotti/ixxatcan
Can it be integrated/ used?
Thanks

rusoku, 2022-11-21:
It should work but up to 5.15 ver only.

pink-fr3ud, 2022-11-21:
Is this because of QT6?
I won't be able to try this, my programming skills are not enough to merge two codes and compile a binary 
:/

rusoku, 2022-11-21:
Why do you need Qt6 version ? Use Qt5.15.2 for both.

pink-fr3ud, 2022-11-21:
Sorry we misunderstood each other
You say it will work up to 5.15 because it is the last revision of QT5 ? And until savvyCAN and the plugin go QT6 it's the end of the road?
Anyway I don't have the knowledge to do it myself yet, I have some reading to do
Thank you

rusoku, 2022-11-21:
I compiled ixxat plugin and SavyCAN Qt5.15.2 & Win11_64 bit version.SavyCAN find plugin but I don't have ixxat hardware to test it.

[image]

pink-fr3ud, 2022-11-22:
That was quick... Can educate me how to do it or share the binary ?

rusoku, 2022-11-23:
Both plugin versions: SOCKET and IXXAT OLD
https://www.rusoku.com/downloads/5.15.2_win64_IXXAT_plugins.zip

pink-fr3ud, 2022-11-23:
It worked flawlessly, I'm able to listen and send data to the bus ! Do you want me to try something with this hardware?
Thank you!