collin80/SavvyCAN #546: Better support for CAN-FD

collin80, 2022-11-29:
When SavvyCAN was originally written nobody used CAN-FD. Fast forward many years and now FD is becoming more common. Many parts of the backend have already been changed to support CAN-FD but many of the GUI portions have not. Throughout the program the 8x8 CAN Grid GUI element is used.  This obviously only supports 8 data bytes each of which has 8 bits. CAN-FD can have up to 64 data bytes. A 64x8 grid would be too crowded. It may be possible to divide this up a bit better. Maybe a 32x16 grid could be viable. Here is a quick mockup of where the divisions would be for a 32x16 grid taken directly from a screenshot of the FlowView window so this is a 1to1 actual size that the grid could be.

[image]

To me, this looks potentially viable. The biggest issue is that each grid cell could no longer be labeled as is currently done.

cdxa1, 2022-11-29:
But if FlowView grid will be scalable it can be labeled after exapnding window to needed size. More over, grid cells can be changing according to windows size 32x16 in start view with limited labeling, up to 64x64 with full labeling. With modern displays resolution that shouldn't be a problem.

anonymouslojz, 2023-12-25:
@collin80 hi all, we innomaker already update the savvy software to support can fd, you can try and test with it. We including our innomaker can and canfd products. Please check from below github link [https://github.com/INNO-MAKER/usb2can/tree/master/For%20Windows/SavvyCAN-InnoMaker](https://github.com/collin80/SavvyCAN/issues/url)

[image]
[image]