collin80/SavvyCAN #548: Program crash if new log is loaded while flowview and/or graph window open

collin80, 2022-12-04:
This isn't a sure thing but there are reports that sometimes the program crashes if a new log is loaded while certain windows are left open. No crash happens if all sub-windows are closed first. The windows most likely to do this are flowview and graph.

collin80, 2022-12-09:
Just a note - I cannot duplicate this. The user who had the problem says it happens multiple times a day but I believe they're loading a lot of files and doing a lot of logging. So, maybe it's a race condition I'm just not hitting yet. Anyone who has had this happen, please explain what you remember about the circumstances and I'll see if I can figure out what is going on.

ElDominio, 2023-01-03:
https://youtu.be/cZqJaBIa3-s

Best I can do, I always crash when opening a new file even if I don't have anything else open

collin80, 2023-01-13:
I could only reproduce this one way so far. If I used the "Frame Info Window" then loading a new file after closing that window would indeed crash the program. But, so far I cannot get it to do this if I use any other window. The fix for that particular crash is 0d8a35c7d51de93880b020d9e03f9e653f90e388 but I'm not at all confident that this will do anything to stop any other crashes. The original reporter of this issue uses MacOS and I can see you're using Windows. I have tried to reproduce this on MacOS and couldn't. My main development machine is Ubuntu Linux but I'll try it in Windows 10 and see if that somehow creates any change. It's always frustrating for everyone when I can't reproduce issues that multiple people are having but eventually we should narrow this down to whatever is going on.