collin80/SavvyCAN #556: ixxat device connecting to savvycan

keuninkske, 2022-12-26:
hello,

i would like to connect my ixxat device (usb-to -can V2 compact to savvycan
this because savvycan gives me more reverse engineering possibilities then the ixxat device provided software (minimon) gives me

the ixxat device is not emulating a com port
what are my possibilities to establish this connection?
thanks for your help

rusoku, 2022-12-26:
Did you try this plugin ?
https://github.com/collin80/SavvyCAN/issues/542

keuninkske, 2022-12-26:
hello, thanks for you info
havent found this one ....

now i downloaded https://www.rusoku.com/downloads/5.15.2_win64_IXXAT_plugins.zip
i can find 2 DLL files there
what should i do with them?
to which folder should i add them?

rusoku, 2022-12-26:
There are two version of plugins.You need to install into can bus plugin directory.

keuninkske, 2022-12-26:
got it working, thank you

the DLL file qtixxatcanbuss.dll should go in the /canbus folder of the savvycan program

then you can find the device under qtserialbus devices

you guys are the heroes of my day
thanks allot!!

keuninkske, 2022-12-26:
solved

closed 2022-12-26 by keuninkske