collin80/SavvyCAN #561: Bug in DBC File Editor?

UlrichThiess, 2023-01-04:
Version: 2.08 Windows

In the DBC File Manager i load the file CSS-Electronics-OBD2-Extended-v1.4.dbc from CSS Electronics.
I saved it as Test.DBC.
If i reload this file, i get some errors:
[image]

What ist my mistake?

collin80, 2023-01-06:
Hmm, not sure what the problem is. I just loaded that file and there were no errors. Then I saved it to test.dbc and reloaded it. That worked too. Granted, I'm not in Windows and using a pre-release version but as far as I remember, nothing has changed in DBC loading since the published V208. Just to be sure, can you attach the resulting test.dbc file you created so I can try that to ensure there wasn't some quirk I'm not seeing?

collin80, 2023-01-06:
I take that back. I did make a number of changes to the DBC code and some of them may have fixed whatever issue is causing it to tell you so many signals are faulty. You should try the latest continuous build to see if things improve.

UlrichThiess, 2023-01-07:
Load - Save - Reload works with the newest Developer Build. Thanks.

But sadly this file could not load with Vector CANdb++ Editor:
[image]

Also in CSScanedge-influxdb-writer i get errors:
[image]

I use a CANedge2 to push the CAN Data to my own Server. I convert MF4 files into ASC. Look at it with SavvyCAN, make my own DBC File and then i put the data into an InfluxDB with help of the DBC File. THE CANedgeInfluxDBWriter only works with DBC Files made with Vector CANdb++ Editor.

collin80, 2023-01-07:
Hmm, unfortunately that's a very common problem. I recently saw discussion over at OpenDBC where people were saying that the OpenDBC files don't load properly in CANdb++. 

I'm not 100% sure why the errors are happening. The output from CCScanedge-influxdb-writer might have a clue though. I see the \n on the end of those lines. I'll bet the standard probably wants only /r as a line ending but a lot of us are using \n instead. Normally text editors support all the line endings, \r, \n, and \r\n but maybe that python script is picky. As for the CANdb++ error, I'm not sure. It doesn't seem to be giving the full line and I don't know for sure what line 200 is. When I tried the same thing you did I got a file that has this as line 200:

SG_MUL_VAL_ 2564485392 S1_PID_01_MonitorStatus ParameterID_Service01 1-1;

I'm pretty sure I did the same things you did. But, I don't see the problem with that line. I may have to boot up Windows and install a trial of CANAnlyzer so I can get CANdb++ and test this myself.

mitchdetailed, 2023-01-07:
candb++ is free, ill get you a Google drive link to avoid the headache of
navigation ...one moment

mitchdetailed, 2023-01-07:
https://drive.google.com/file/d/1hVSgIwxZEOF15moh0ez2z_gpWS21JBHF/view?usp=share_link

UlrichThiess, 2023-01-07:
Thanks for your hint. Yu're right:

CANdb++ Lineending: 0xd 0xa
Your Lineending: 0xa

Could you change this?

UlrichThiess, 2023-01-08:
Sorry, but thats not the Problem. Even if i change the line endings into DOS with unix2dos it won't work in type python script from CSS Electronics nore in CANdb++ Editor.

FiatDucato.dbc is the original File from CSS: CSS-Electronics-OBD2-Extended-v1.4.dbc with two additions edited with CANdb++. It works with the scrip and with SavvyCAN.

Test.dbc is FiatDucato.dbc loaded and saved with SavvyCAN 2.09. Test.dbc could loaded by SavvyCAN without errors. Test.dbc could not load by CANdb++ nor with the CSS canedge-influxdb-writer python script.

Both files has DOS Line endings. Everything was made with Windows Programs.

[FiatDucato.dbc.zip](https://github.com/collin80/SavvyCAN/files/10368081/FiatDucato.dbc.zip)
[Test.dbc.zip](https://github.com/collin80/SavvyCAN/files/10368082/Test.dbc.zip)

collin80, 2023-01-11:
I committed a fix for all the issues: 711d5e886035e6723fb873377477cda109bfe843

I have tested your FiatDucato file to still load in CANdb++ after having been saved in SavvyCAN. Hopefully this squashes all the bugs but time will tell.

UlrichThiess, 2023-01-11:
Perfect. Many thanks. You did it.

Now it works with CANdb++ and with the CSScanedge-InfluxDB-Writer Python script from CSS Electronics.

closed 2023-01-11 by UlrichThiess