collin80/SavvyCAN #562: GVRET Serial Connection Issue on OSX

jmeyer-fluid, 2023-01-05:
V208 fails to connect to my ESP32 CANDue over serial on V208.  I'm running SavvyCAN on a M1 Pro Macbook, OSX version 12.3. 

I've tracked the issue down to PR #484, the change to default to a TCP based connection if a `.` is in `pPortName` in `canconfactory.cpp` causes the issue for me.

The GVRET device shows up as `tty.usbserial-ABCD` on my machine (replace `ABCD` with some numbers, I'm guessing these are related to the USB port chain the device is connected to, as the number moves around depending on what port I'm connected to and if I'm using a hub). 

I've found an easy workaround by changing the `if(pPortName.contains(".")` check to `if(pPortName.contains(".") && !pPortName.contains("tty"))`. I'm not sure if this will cause a regression for @Sheimy's original change though. What does `pPortName` look like for issue #484 was trying to correct?

collin80, 2023-01-05:
Ahhh, I can see how this would be a problem. Strange, I'm sure I just recently checked things with a Macbook Pro and it seemed to work. Maybe I was using a board with a different processor that presents itself a bit differently. I'll get this fixed within the next day or two.

collin80, 2023-01-06:
Your suggested change has been merged. They were trying to solve a problem where it would load the driver and was supposed to input an IP address but it would get interpreted as serial and not work. So, the check for a dot character was to see if it was an IP address. They don't contain tty so this fix should make both situations do the proper thing. Merged as commit 4e2c52c7a02b461f427dc3743f23f5ebd97117a6

closed 2023-01-06 by collin80