collin80/SavvyCAN #563: J2534 (passthrucan) connection does not stay connected #208

collin80, 2023-01-06:
### Discussed in https://github.com/collin80/SavvyCAN/discussions/560

<div type='discussions-op-text'>

<sup>Originally posted by **Tmod700** January  2, 2023</sup>
I got x64 Build 208 to show the port selections when choosing a passthru connection but it connects and then disconnects itself just a few seconds later. I am using Tactrix Openport 2.0 and Win10x64 and downloaded the 208 CIBuild.
I tried 207 compiled for 32bit and it connects and stays connected fine.
Any ideas as to why it auto disconnects?
[image]
</div>

Bill400, 2023-01-20:
Using Visual Studio 2022 to examine the V208 source files.
Connections module\Canbus.cpp appears to contain our default bus speed of 250Kbts. Statements that follow appear to set the operating parameters. Not yet the C++ expert, I wonder about the last two statements, which are identical except for the const keyword which specifies that the object or variable isn't modifiable. Is the final statement without the const keyword an error? Should it have been deleted?
Further, a work around (for 500Kbts) might be to change the default 250000 to 500000?

amiralnar, 2023-03-14:
> Hi guys I have OpenPort 2.0 clone from China and I was able to compile 32bit version with hardcoded 500kb/s speed for "passthrough" connection and it worked like a charm while i send debug frames from my desktop PC. But the same compiled binary running on my laptop behaves exactly like Collin said - connection drops in a few seconds after it was "Connected". Both devices are running Windows 10 x64, both have the same 32bit OpenPort driver. I feel the main difference is that laptop have no Qt SDK installed and binary uses some included lib. Anyway i'll try to investigate that deeply and who knows maybe i'll find a solution for that
> 
> EDIT: Nevermind. I just reinstalled OpenPort driver (i believe exactly the same driver was installed, strange thing) and now my adapter remains connected. Now it fully covers my needs, Thank you for a great application!

How did you hardcoded speed exactly? @macbaze

dmg210, 2023-11-19:
@macbaze Any chance you could share the complied 32bit version with the 500kb/s hardcoded? Thanks.

amiralnar, 2023-11-20:
> @macbaze Any chance you could share the complied 32bit version with the 500kb/s hardcoded? Thanks.

Try this one: https://github.com/amiralnar/SavvyCAN/releases/download/continuous/SavvyCAN-Windows_x32_CIBuild.zip

dmg210, 2023-11-20:
Legend! Thanks. That's working a treat with the Tactrix OpenPort 2.0 👍

mrdka69, 2024-12-31:
Did you guys connect pin 6 and 14 of the openport 2.0 to the can H and can L of the given device? Im trying to do the same byt unfortunately dont get anything in savvycan. Ive downloaded the savycan from @amiralnar

dmg210, 2025-01-01:
Yeah I just used the regular obd pinout on the openport. Pin 6 - can high, pin 14 - can low.

rovis-hub, 2026-02-07:
В новій версії взагалі не бачить прилади по passthrucan. Збірка розробки QT6 [https://github.com/collin80/SavvyCAN/releases/tag/continuous-QT6](url)