collin80/SavvyCAN #571: CSV file timestamp incorreect/negative

TripleFJeff, 2023-01-30:
I have a saved CSV where the timestamps are negative. If it is loaded back in they all show up as zero. If I go into the CSV and remove the negative sign it loads fine.
[TPMS Log.csv](https://github.com/collin80/SavvyCAN/files/10531292/TPMS.Log.csv)

Insectodium, 2023-03-08:
We are experiencing the same issue
SavvyCAN V208 for Win10

collin80, 2023-03-09:
Hmm, that's strange. SavvyCAN saved the file like that with negative timestamps? Obviously that's not something it should be doing. I see a couple of potential clues.

1. They really are in reverse order if you just remove the negative sign. So, they were in proper order before that but have a huge negative offset
2. If you do remove the negative signs and load it then interpreting the time stamps as "System Clock" shows that the capture was done on 29-Jan 2023 at around 9:48am which I assume is accurate.

So, this seems like a glitch in applying a time offset for keeping track of the current system time. It seems like it gets applied backward and creates a very large negative offset. Which capture device was used for the file? Maybe it's a bug in one of the capture drivers. And, were you using system clock mode for timekeeping? With the answer to these questions I should be able to find the source of the problem and fix it.

TripleFJeff, 2023-03-09:
I was using a PCAN. 

The value in Byte 1 of my file should count up from 1 to 4 and repeat, which it does in the csv file so i know it's in the right order. If I simply remove the negative sign and resort, it counts down from 4 to 1, so it is indeed a big offset rather than simply just having a negative attached to the correct time stamp.

 The file was recorded on Jan 29th. The created date of the CSV is 3:12PM, I don't recall what time it actually was but the date is right.

TripleFJeff, 2023-03-09:
One other note-- its not consistent. I've only seen it once and other captures from the same setup were fine.

Insectodium, 2023-03-09:
We do have it in all captures for the last 4 months. On two different vanilla installs on two different Win10 laptops.

cschweers, 2023-03-09:
32 Bit or 64 Bit Win10?

Insectodium, 2023-03-09:
Win 10 Pro 64bit on at least one of them, unsure on the other...

bbyers-UBCO, 2024-08-26:
I am seeing this, was there ever any resolution here?