collin80/SavvyCAN #572: Cant choo se

spot2000, 2023-01-31:


spot2000, 2023-01-31:
When trying to use the UDS scanner, it is not possible for me to choose an extended ID as starting ID, only for the ending ID.

So if I want to scan between two extended ID then its not possible? or is it a bug?

[image]

spot2000, 2023-02-04:
Anyone else with the same problem, is it by design or a bug in th UI?

spot2000, 2023-02-04:
Do you recognize this problem @collin80 ?

collin80, 2023-02-06:
I haven't had a chance to look at it yet but I'm sure I failed to test with extended IDs so there is very likely some issues to resolve. I'll try to get this fixed up soon.

spot2000, 2023-02-14:
Hi @collin80. Now I looked at the code again but I cant find it (probably dont understand the code), did you have any time to look at this matter?

collin80, 2023-02-15:
I looked into it. The problem was that the starting address is never allowed to be larger than the ending address. So, if you input the starting address first it will stop you from entering a value larger than the ending address. If you instead input an extended address into the ending address then go back to the starting address it will then let you type in an extended address so long as it is not larger than the ending address. But, as evidenced by this becoming a problem for you, this is not obvious or clearly communicated in any way. Here is what I'm going to do: if the starting address is larger than the ending address then I'll just make the ending address the same as the starting address automatically. This then satisfies the requirement that starting address is not larger than the ending address.

This problem is fixed as of 47bae6d65889243179d508688d8635ace9976c73

Now, you can go ahead and set the starting address first. Any time there is an inversion it'll just force the ending address to be at least the value of the starting address. This has also been implemented for service and sub-function.

closed 2023-02-15 by collin80

spot2000, 2023-02-15:
Thanks for your support @collin80