collin80/SavvyCAN #574: SavvyCAN crashing on start

MrDIYca, 2023-02-02:
Love the app!

Yesterday, I started having an issue where the app crashes on start. I haven't been able to use it since. The only thing I remember changing was the 'Maximum Frames to Capture' in the settings. I x100 it . I don't know if this is relevant or not. I have no way to changing it back due to the crash on start issue. 

I tried to see if the app saves its settings in ~/Library so I would delete it to reset it, but can't find any.

Attached to the ticket is the full crash log.

> Process:               SavvyCAN [10367]
> Path:                  /Applications/SavvyCANs.app/Contents/MacOS/SavvyCAN
> Identifier:            com.yourcompany.SavvyCAN
> Version:               ???
> Code Type:             X86-64 (Translated)
> Parent Process:        launchd [1]
> User ID:               501

> Date/Time:             2023-02-01 21:35:03.9697 -0500
> OS Version:            macOS 13.1 (22C65)
> Report Version:        12
> Anonymous UUID:        85037995-4EF1-50F3-FD90-E2ED5141CAED

> Sleep/Wake UUID:       4CADCCA1-7D9C-4946-94EC-517AC962ED6F

> Time Awake Since Boot: 30000 seconds
> Time Since Wake:       5074 seconds

> System Integrity Protection: enabled

> Crashed Thread:        0  Dispatch queue: com.apple.main-thread

> Exception Type:        EXC_CRASH (SIGABRT)
> Exception Codes:       0x0000000000000000, 0x0000000000000000

> Application Specific Information:
> abort() called

[crash.log.txt](https://github.com/collin80/SavvyCAN/files/10564286/crash.log.txt)

MrDIYca, 2023-02-12:
Found it:  

`/System/Volumes/Data/Users/$USER/.config/evtv.me/SavvyCAN.ini`

Deleted and all is fine now.

collin80, 2023-02-14:
I'm glad you got it solved. The problem, as evidenced by your crash dump and the change you made, was that it tried to allocate too large of a buffer and could not do so. The default is 10 million frames. For efficiency reasons, whatever value you put into the "Maximum frames to capture" field is what it tries to pre-allocate so that the buffer never needs to be resized. Resizing a buffer of 10M frames would be an unpleasant experience as all existing frames have to be copied after a larger buffer is allocated. Each CANFrame object allocated is 56 bytes during the pre-allocation. So, 10M is already half a gigabyte of RAM. Asking for 100x more is 50 gigabytes of RAM which is probably more than you have. Thus the operating system panics and the program dies.

There are valid reasons why it works the way it does but I should add some sanity checks to make sure that nothing crazy is happening. You had no way of knowing that it would try to literally pre-allocate all that RAM so it's not your fault. It should probably be made more clear how things work and also error check the allocation so that it handles things gracefully if the allocation is impossible.