collin80/SavvyCAN #577: To do: Port to Apple 'M' series SOC

exekutive, 2023-02-13:
Compile to run natively on new current Apple hardware.

collin80, 2023-02-14:
This should be possible with the CI builds. I believe that XCode can cross compile to the Apple Silicon hardware. I'll have to check if QT 5.15 really supports it though. This may require QT6 which SavvyCAN has not fully been ported to yet.

rusoku, 2023-02-17:
Apple Silicon arm64 version, Qt5.15.8, Apple SDK: macOS 13.1 
https://www.rusoku.com/downloads/SavvyCAN.dmg

rusoku, 2023-02-17:
Version with Qt CAN bus plugins:
http://www.rusoku.com/downloads/SavvyCAN_arm64_plugins.dmg

rusoku, 2023-02-17:
commit: 6405841, arm64 Apple Silicon
http://www.rusoku.com/downloads/SavvyCAN_arm64_ 6405841.dmg

hhglae, 2023-08-13:
Its working fine on aarch64. You just have to compile it instead of using precompiled github binaries. I am using Arch Linux with https://aur.archlinux.org/packages/savvycan atm.