collin80/SavvyCAN #583: CAN GATEWAY DOES NOT WORK ON ESP32

Magneelektronics, 2023-02-17:
Hi, I used savvycan with the SAM3X8E microcontroller, it was possible to combine CAN0 with CAN1 at different speeds and it worked as CAN gateway, now I use esp32 CAN0 and CAN1 on MCP2517FD works great but does not connect two CAN like CAN gateway, is there a solution for this to work as on the SAM3X8E

collin80, 2023-02-21:
Well, how were you doing that before? SavvyCAN isn't exactly meant to act as a CAN bridge but there are a variety of ways such a thing could be done within the program. I'm not aware of any particular reason it shouldn't work the same between the two different devices.

Though, it might be interesting to add a mode more specifically geared toward doing CAN bridging. If such a mode supported an active filtering interface where frames could be selectively let through one way or the other then that would be useful I think.

Magneelektronics, 2023-02-21:
The CAN gateway works immediately on the GVRET firmware, just send can ca CAN0 data and they are automatically sent via CAN1 and vice versa, it is a fixed speed of can 500kbs if we are not connected to savvycan and we do not choose other speeds, but if we want the gateway to work as a gateway it is enough to set different speeds, for example CAN0 500kbs and CAN1 250kbs and the gate works properly, data sent to CAN0 with a speed of 500kbs, are automatically sent to CAN1 with a speed of 250kbs, it is a very good option but unfortunately it does not work on ESP32RET, I also use CAN analyzer hacker and it is optional to enable the gate and filter data from channel 1-2 2-1, we can test it on ESP32 if you want I can work on it, I think it will be a very good option for the gateway in savvycan also on ESP32[]([image])
[image]

Magneelektronics, 2023-02-21:
[
[image]
](url)

collin80, 2023-02-22:
ahh, yes, GVRET did have that functionality. At some point in time someone wanted that and they implemented it by making it automatically do forwarding between the two buses if a given pin was high. The pins used for this are 11 and 12. Since the code defaults to pulling both pins high you end up getting the forwarding by default. That's not exactly ideal. We shouldn't have defaulted to that condition. I wonder if that has caused troubles for various people. I will have to fix that so that forwarding isn't the default condition. You could still get the desired behavior by pulling pins 11 and 12 to V+. But, really there should have been a software based way to set this. GVRET has been set in stone for many, many years with no updates. But, I can fix that. Anyway, back to ESP32RET, I can add the functionality to ESP32RET as well but I'll be doing that with settings you can trigger from the serial console instead of pins you pull high or low. I am also adding the functionality to SavvyCAN which will allow forwarding between two buses on different hardware interfaces.

Magneelektronics, 2023-04-11:
Hi coolin80, is it possible for you to add gateway capability to savvycan and modify the ESP32RET? should I add a code to ESP32RET to activate the gateway in the same way as GVRET giving a low or high state to some ports of the ESP32 processor

collin80, 2023-04-11:
It was added to SavvyCAN with commit 73d7e4d25465d085428e771c313dbf61fafcb827 which happened back in February. If you download the latest continuous build you should have a CAN bridge window that will give you the functionality you seek. I still haven't added it to ESP32RET.

Magneelektronics, 2023-04-12:
yes sorry I didn't notice that you added :) and have you tried to add it to the ESP32RET firmware, is there a big problem with it? I will try to modify the ESP32RET code but I don't know if I will be able to integrate it with the CANBRIDGE option with SAVVYCAN, I could do it like in GVRET but I'm afraid it won't work