collin80/SavvyCAN #599: Change X-Axis timestamp

Lukasusa, 2023-03-14:
Hi,

is it possible to change the timestamp on the X-Axis in the Data Graphing Window?
For example to see the hours and minutes.

[image]

collin80, 2023-03-15:
Well, sort of. If you set your main time keeping mode in the preferences window to seconds then the graph will be labeled in seconds instead of microseconds. This makes it a little bit better but in your case it looks like your captured values are probably actual timestamps with the full date and time. This doesn't ever (at the moment) get respected in the graphing window so you'd always see seconds since the epoch (1970?) You can make the values look a little nicer by using the "Normalize Frame Timing" button on the main screen but that destroys your timestamps. I'll categorize this as an enhancement request. The fact is, I was always capturing with hardware that set timestamp 0 as the start up time of the capture device so the way it works was fine for me. Now a lot of people are using hardware that captures the actual timestamp and this doesn't work quite as well in some spots. So, I really ought to fix it.

collin80, 2023-03-15:
I just pushed an attempt to make this better: bfa96f428e6a4dcf7c1d120d69c8696a3cbea849

Now the main graphing window should show the X axis in hours:minutes:seconds.milliseconds since the start of the day. In general this should do the right thing but it won't if a capture happens to span across a day boundary. If needed it can be switched to day/time format but this is really large and I'd rather avoid it if possible.

Lukasusa, 2023-03-16:
Thanks for your quick solution.
It works great.

closed 2023-03-16 by Lukasusa