collin80/SavvyCAN #600: Adding timestamps?

lansalot, 2023-03-17:
Am this library with a Nano, MCP2515 and SavvyCAN, but I can't get timestamps from it:
https://github.com/latonita/arduino-canbus-monitor

I thought setting System Clock in Preferences would inject the capture time according to PC rather than dongle, but it's still blank - am I doing something wrong?

I tried sending Z and Z2 commands via SavvyCAN text sending, but that didn't seem to help...

What am I missing?

Thanks

Dreed420, 2023-03-18:
Saavycan and lawicel working On Windows?

lansalot, 2023-03-18:
Yes on Windows, but I got same on linux

Dreed420, 2023-03-18:
I cannot send or recive any mesaages on Windows (saavycan)

amiralnar, 2023-03-19:
@lansalot > What am I missing?

For now the Lawicel protocol implementation uses timstamps only from device and the problem is that the arduino device dosen't provide timestamps unfortunately.

@Dreed420 > I cannot send or recive any mesaages on Windows (saavycan)
It's because of using serial connection with Hardware conrol option.
Try to create connection, then remove it and then create again. It can force bytes thru serial without hardware control. It works for me. Also i use the Terminal 1.9 program to start CAN flow from device by sending commands manual: C -> S6 -> O and after starting the flow from device i create connection in the program. Then it can recive messages.
Buh it can't send messages anyway.

@collin80 
Maybe you can add the Hardware flow control option and the timestamps from the system?

lansalot, 2023-03-19:
Thanks @amiralnar 

I might see if I can just inject millis() into the Arduino lawicel library, but it would be handy if savvycan could just fill it in if blank of course.

Is that what your custom build does ?

amiralnar, 2023-03-19:
> Is that what your custom build does ?

Yes it uses system time timestamps 
And it uses serial port without hardware flow control option. This is useful for some Arduino devices which has only RX TX and GND signals.

lansalot, 2023-03-19:
Throwing "the application was unable to start" straight away, it's in same folder as all the support code.

lansalot, 2023-03-19:
Went back to the library source, and if I hard-code in the addition of timestamps, and enable _MCP_FAKE_MODE, the sample format changes from

t10440DC90000
t10640BD80000
t105409AA0000
to
t10F40DBF00000D36
t10740BC600000D37
t10740BA50000A32F

So I can see the 4-byte extended mode timestamps. I also tried with 2, but savvycan still not reading them?

amiralnar, 2023-03-19:
> Throwing "the application was unable to start" straight away, it's in same folder as all the support code.

It's 32bit exe file. 
Maybe you trying to place it in 64bit program folder?

amiralnar, 2023-03-19:
> Went back to the library source, and if I hard-code in the addition of timestamps, and enable _MCP_FAKE_MODE, the sample format changes from
>
> t10440DC90000 t10640BD80000 t105409AA0000 to [...]

Wow, you did a great work! Sorry i said it wrong.
There is no capture the timestamp in SavvyCAN for now at all
Take a look at https://github.com/collin80/SavvyCAN/blob/master/connections/lawicel_serial.cpp 451
You need your own build anyway for now.

amiralnar, 2023-03-20:
@lansalot now you can try to use full dev release from my fork: https://github.com/amiralnar/SavvyCAN/releases

lansalot, 2023-03-22:
Thanks, heading out to farm soon, will give it a try !

lansalot, 2023-03-31:
Worked a treat, thanks!

lansalot, 2023-03-31:
Altho, the codes savvycan retrieved weren't the PGNs I was expecting. Truncated to 4 bytes if I remember right? Expecting, for example. 0x18ff4f23

amiralnar, 2023-04-01:
> Altho, the codes savvycan retrieved weren't the PGNs I was expecting. Truncated to 2 bytes if I remember right? Expecting, for example. 0x18ff4f23

Pls explain more detail. Didn't get it

lansalot, 2023-04-03:
I think it's because the middle 2 bytes are being displayed by SavvyCAN, but I see all 4 in Canhacker for example.

eg 0x18ff3423 would show as 8ff3 in SavvyCan?

amiralnar, 2023-04-03:
> I think it's because the middle 2 bytes are being displayed by SavvyCAN, but I see all 4 in Canhacker for example.
> 
> eg 0x18ff3423 would show as 8ff3 in SavvyCan?

Can you show me the screenshot example from canhacker and from savvycan?

lansalot, 2023-04-28:
Ah sorry, I was meaning the 29-bit ID. Apologies for delay in reply, changed jobs, life hectic etc.

Still getting connections only occasionally with SavvyCAN but CanHacker always seems to connect, no idea why?

closed 2023-08-27 by lansalot