collin80/SavvyCAN #605: Bug: Playback using 'manual' playback speed seems to not work

MatinF, 2023-03-23:
**Observed**
When using the playback in v213, the data is replayed as expected when using the original frame timing (tested with a CLX000). 

However, if the 'manual' playback is selected, the `Playback speed (ms)` does not seem to properly work. It seems to cause frames to be output at unexpectedly slow or arbitrary rates.

**Expected**
If I set the playback speed to e.g. 5 ms, I would expect to see the selected messages being output at a 5 ms frequency. 

See below video and data for context

https://user-images.githubusercontent.com/26184427/227222462-62558c24-1f69-4425-b546-8e6cdf976600.mp4

[file_for_replay.TXT](https://github.com/collin80/SavvyCAN/files/11051458/file_for_replay.TXT)

collin80, 2023-03-24:
The problem would seem to be asking for precision timers from the operating system. This is an easy thing on an embedded system. You can ask for timers with microsecond precision on little 40MHz 16 bit processors. But, for whatever reason you ask a quad core 3Ghz processor running Windows to give you a 5ms timer and it just doesn't reliably. It seems Linux is giving me reliable 5ms ticks as I can ask for that and get 200 frames per second with fairly decent reliability - it ranges from 197 to 201 frames per second. But, as I get toward 3 or 2 ms it starts to break down a bit.  "Use original frame timing" actually asks for a 1ms timer but keeps track of how long the actual duration is and sends any messages that should have been sent in the duration since last tick. Since it tracks the actual timing it tends to trend toward the proper number of frames per second. It is possible that I may need to do something similar to get better accuracy of the normal "send at interval" mode. The current accuracy is very dependent on whether the OS schedules a tick back to the playback system when I ask it to. That's not a sure thing on any operating system.

amiralnar, 2023-03-24:
@MatinF Hi! Please show me your connection properties.
I have similar issue when i send CAN messages via direct COM port connection. But the same messages goes pretty good via QTSerialBus Devices like passthru.

rusoku, 2023-03-24:
I moved all delays between can bus frames to adapter and modified a SavyCAN app and qt plugin to send only info about delays.
The adapter continues to perform its work with any interframe resolution.

MatinF, 2023-03-24:
You can see the steps I take to connect our CLX000 via the video at the top of below link:
https://canlogger.csselectronics.com/clx000-intro/log-file-tools/savvycan/

@collin80 What is strange is that in a previous version of SavvyCAN (release v207), we had the opposite experience. Here, the 'original frame timing' did not work properly, but we were able to use the manual frame timing, see e.g. below video:
https://canlogger1000.csselectronics.com/img/CAN-bus-sniffer-playback-log-file_v2.mp4

It should also be noted that we have no issues sending multiple frames at high rates (e.g. 1 ms or 5 ms intervals) onto the CAN bus when using the 'Custom frame sender' tool. So my thinking is that this relates to a difference in how the manual playback is implemented vs. e.g. the custom frame sender.

MatinF, 2023-03-24:
Hi again Collin,

I reviewed this further:
1) I believe both the original frame timing and the manual frame timing work as they should, I was incorrect
2) What caused my confusion was that if the loaded CAN IDs are filtered (e.g. to show only 1 CAN ID), I would have expected that this single CAN ID would be output at the manual frame timing (e.g. 50 ms). However, instead, it seems the 50 ms is applied across all the CAN IDs - so if I e.g. have 10 CAN IDs with equal frequency in the original data, I would be seeing only my selected CAN ID - but the rate at which it arrives would be 10 x 50 ms instead of every 50 ms.

It might be worth considering if the 50 ms should be applied "after" the CAN ID list is filtered as I think that might be more intuitive. But since this is more nitpicking I will close this issue for now.

closed 2023-03-24 by MatinF