collin80/SavvyCAN #608: Feature: Include existing plugin for CLX000/CANmod support in Master

MatinF, 2023-03-23:
Hi Collin,

We provide a plugin to our users for SavvyCAN that enables integration of our [CLX000](https://www.csselectronics.com/pages/can-bus-hardware-products) (CAN bus data logger & interface) and [CANmod](https://www.csselectronics.com/pages/can-bus-hardware-products#sensor-to-can) (sensor-to-CAN series. The plugin is attached here and simply needs to be put in the `canbus/` folder:

[csscan_qt5.zip](https://github.com/collin80/SavvyCAN/files/11055033/csscan_qt5.zip)

It would be great if you can include the plugin in upcoming releases of the master branch. This will simplify the process for those of our users that wish to always use the latest version of SavvyCAN.

SavvyCAN is used by many hundreds of our end users and we're very happy for all your great work!

collin80, 2023-03-24:
Thanks. I'll get it packaged into the windows build, both for continuous releases and any subsequent actual releases.

MatinF, 2023-03-24:
Awesome Collin!

miguel-asti, 2024-08-23:
Can I request for the csscan.h and csscan.cpp files so that I can also use it on linux? I have a CANmod.router and I can only use it on windows as of the moment. Thanks.

MatinF, 2024-08-26:
We will get back regarding this by email Miguel

closed 2024-08-26 by MatinF