collin80/SavvyCAN #609: Feature: Detach

MatinF, 2023-03-24:


MatinF, 2023-03-24:
**Observed**
In the current version of SavvyCAN there are two pain points related to the trace window and buffering of historical data:
1. If a user enables the 'Overwrite' mode, all historical data is lost
2. If a user has enabled the 'Overwrite' mode and tries to create a Graph of the data, nothing is shown

We assume both of the above relate to the fact that the buffering/storing of historical data is somehow directly linked to the trace window in which it is displayed in the GUI. This might be done originally to keep the implementation simple, but has the above downsides.

**Suggestion**
If possible, it would be ideal to store the historical data separate from the trace window. This data can then be displayed in the trace window, but the user can freely jump back and forth between the trace view and overwrite view because it has no implications on the underlying data source. 

Further, this would also allow users to watch their data in the overwrite mode, while at the same time creating graphs of the data.

amiralnar, 2023-03-25:
Hi! Maybe the sniffer window can solve it?

MatinF, 2023-03-26:
> Hi! Maybe the sniffer window can solve it?

I am not sure how this would help in regards to this issue?

amiralnar, 2023-03-26:
> > Hi! Maybe the sniffer window can solve it?
> 
> I am not sure how this would help in regards to this issue?

I think in this case you can turn off overwrite more and save frame history and see the last data in sniffer window at the same time

MatinF, 2023-03-26:
Ok, but I think this is more of a fundamental point that I think savvycan would benefit from disconnecting the historical trace data from any specific GUI window, thus enabling users to eg freely toggle the mode for which they view data without causing eg the clearing of historical data