collin80/SavvyCAN #611: Added support for a TouCAN adapters from Rusoku technologijos

rusoku, 2023-03-25:
Released Qt 5.15.x open source serial bus (CAN bus) plugin for TouCAN adapter and Apple MacOS.
It supports up to eight adapters at the same time.
Any serial port devices ;-)
[image]

mickeyl, 2023-05-14:
@rusoku Excellent. Is this already or will this appear in Qt 6?

rusoku, 2023-06-11:
Released Qt 5.15.x open source serial bus (CAN bus) plugin for TouCAN adapter and Windows x64.
Plugin currently supports up to eight devices.
[www.rusoku.com/products](https://www.rusoku.com/storage/app/media/drivers/Windows/qtrusokucanbus.zip)

[image]

closed 2023-08-05 by rusoku