collin80/SavvyCAN #615: SavvyCAN connecting to bus crashes it, but Canhacker(sorry!) fine?

lansalot, 2023-04-02:
Evening all

Discovered SavvyCan and it's awesome, so much nicer than anything else I've seen around. But I've ran into a problem. I'm using a Nano and an MCP2515 shield, and a [Lawicel ](https://github.com/latonita/arduino-canbus-monitor )library for SavvyCan, and  common Canhacker library for Canhacker.

On my Navarra, either config will connect to the vehicle at 500k and it's fine, I can read the codes.

However, at 250k, on my friend's Case CVX tractor, CanHacker was fine and read the codes - but as soon as I tried to read anything with SavvyCan at 250k, it either hung the bus resulting in error codes when I turned the ignition on, or it saw nothing at all. Tried with a terminating resistor and not, just in case, but nothing.

Am at a bit of a loss - I tried amending the Lawicel library just in case it was starting out at 16mhz (my MCP2515 is 8mhz only) or something, but couldn't get anywhere.

As I'd mentioned in #600 , I had been looking for SavvyCAN to inject timestamps, and @amiralnar kindly provided a build to do just that, and that worked great. But if it won't read the data live, it makes things a little more difficult.

Pretty certain I'm just doing something wrong, but have been searching all evening to no avail so far. Any suggestions gratefully received, thanks!

ojuniour, 2023-04-10:
how'd you get this to work with arduino nano?

lansalot, 2023-04-13:
Using the libraries as above?

ghira89, 2023-04-22:
Test Arduino nano with mcp2515 and this libraries but not work ...

ojuniour, 2023-04-22:
> Test Arduino nano with mcp2515 and this libraries but not work ...

I got it working with nano..

ghira89, 2023-04-23:
Help me..? Pls

ghira89, 2023-04-24:
> > Test Arduino nano with mcp2515 and this libraries but not work ...
> 
> 
> 
> I got it working with nano.. 
>
>
>
>help me pls..?

ojuniour, 2023-04-26:
> > > Test Arduino nano with mcp2515 and this libraries but not work ...
> > 
> > 
> > I got it working with nano..
> > help me pls..?

https://github.com/latonita/arduino-canbus-monitor/tree/master/arduino-canbus-monitor

What canbus hardware are you using?

ojuniour, 2023-04-26:
You can also get Arduino nano (with Built Canbus transceiver - MCP2515) 
https://www.tindie.com/products/massivegroup/canduino-v31-atmega328p-with-can-bus/

ghira89, 2023-04-26:
Arduino nano with mcp 2515.. connect sniffing canbus.. but not send message..

ojuniour, 2023-04-28:
> Arduino nano with mcp 2515.. connect sniffing Canbus.. but not send message..

Aaaah Yes. I cannot send message either. I think you may have to change the code a bit to send message. 
There is an alternative though if you only just want to sniff, send, and decode message:
https://github.com/adamtheone/canDrive
here is a video of it:
https://www.youtube.com/watch?v=cAAzXM5vsi0&list=PLZo1kmxBtRyKx3J8CrIkp3CZlgnUTfSR0&index=31

closed 2023-11-04 by lansalot