collin80/SavvyCAN #621: Program crashes when trying to load DBC file

Artur-Romaniuk, 2023-04-18:
Environment:
Linux AppImage
Fedora 37
Release V213 and dev
Bug is not present in V208

Description:
When trying to load a DBC file (DBC file manager >> load) program crashes immediately.  There are no error displayed. Everything works fine using the previous release V208.

StarkJohan, 2023-05-03:
Same issue on Pop!_OS 22.04 LTS with kernel 6.2.6-76060206-generic

collin80, 2023-05-07:
It seems this same thing is reported with Ubuntu 23.04 as well and that person found that reverting to 22.04 fixed the problem. But, I'm using Ubuntu 23.04 and compiling it myself works. Also, I tried the V213 app image and that worked fine too. So, something weird is going on. Can anyone share a DBC file that crashes on this new version? I tried around 10 different files, all of which I thought were pretty complicated and used a lot of DBC features and got no crashes.

nicatec, 2023-06-02:
I am having the same issue on  Linux Mint 20.3, using the current release, as well as SavvyCAN-f397a50-x86_64.AppImage.
If I try to load an existing DBC file, savvycan crashes immediately.
If I create a new DBC file, savvycan holds out until I try to "add a new signal".
Running out of the shell I am getting messages
```
Segmentation fault (core dumped)
and
*** stack smashing detected ***: terminated
```
Both of which result in savvycan terminating.

TadayukiOkada, 2023-07-17:
I'm using Mint 21.1 and trying to load DBC gives this:

```
$ ./SavvyCAN-044fea3-x86_64.AppImage
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgvfsdbus.so
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgioremote-volume-monitor.so
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgvfsdbus.so
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgioremote-volume-monitor.so

(SavvyCAN-044fea3-x86_64.AppImage:26554): Gtk-WARNING **: 22:08:47.020: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Mint-Y/actions/16/image-missing.svg: Unable to load image-loading module: /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so: /lib/x86_64-linux-gnu/librsvg-2.so.2: undefined symbol: pango_attr_insert_hyphens_new (gdk-pixbuf-error-quark, 5)
Aborted (core dumped)
```

TadayukiOkada, 2023-07-17:
I have followed this steps, but replaced libpango instead of libcairo and now I can load DBC:
[openscad/openscad#](https://github.com/openscad/openscad/issues/3523#issuecomment-861524760)
`$ cp /usr/lib/x86_64-linux-gnu/libpango-1.0.so.0* ./squashfs-root/usr/lib/`

LaXiS96, 2024-03-09:
I have the same exact issue as @TadayukiOkada on Pop!_OS 22.04 LTS.
SavvyCAN crashes instantly when clicking on "Load New File" in the RE Tools > DBC Comparison window.

Manual build from source (master) as of today (with Qt 5.15.2) does not have this issue and works as expected.

The error refers to a missing symbol used by librsvg (`pango_attr_insert_hyphens_new`); for reference, the libpango version in my system is `1.50.6+ds-2ubuntu1`. Is the AppImage libpango outdated?