collin80/SavvyCAN #625: Arduino nano / mcp2515

ghira89, 2023-04-24:
I'm looking for help for Arduino nano and mcp2515, one already canhacker but I can't make it work on savvycan..

amiralnar, 2023-04-26:
Arduino must be configured in hardware flow control mode of COM port.

ghira89, 2023-04-28:
Yes read message but not send.. and no set speed can.. using terminal 1.9 for set speed

amiralnar, 2023-04-28:
That's what i've got too. 
I was able to set speed and open CAN connection using terminal and then listen CAN messages.
To fix this issue i had to build my fork with some changes in the source code.
This changes is incompatible with original GVRET adapter, so you may try it with Arduino
https://github.com/amiralnar/SavvyCAN/releases

ghira89, 2023-04-28:
Test program but speed not work.. select speed on terminal 1.9

amiralnar, 2023-04-28:
If you mean bus speed then you have to set it in next window after connection established

amiralnar, 2023-04-28:
And then after some data the connection may be lost. Arduino is very unstable.

ghira89, 2023-04-28:
Work but not possible speed bus select