collin80/SavvyCAN #626: CANable with SAVVYCAN

ufc12345, 2023-04-24:
Hello everyone,

I have a question for you. Can I use SAVVYCAN with CANable? I think ı can do it but ı wonder , Ultimately CANable is a USB to CAN bus converter adapter. 

Thanks for helping.
Best Regards...

nico0481, 2023-05-25:
Hello,

Operating System?

I use it on Linux, in connection parameters, choose QT SerialBus, socketcan and then select your interface, i.e. can0.
You have first to bring up your network connection

Nico

nagup, 2023-06-10:
Also looking to use a CANable dongle with Savvycan. 

**OS: Windows 10**

Its just one I found on ebay, running candlelight firmware so from what Ive read on other issues and google, I should be using SLcan or one of the other options when creating a new issue. It asks for a port but when I plug the dongle in, it doesn't get a COM port at all, just appears in the device manager as a universal serial bus device.

I read a post on the savvycan website where Colin said its not supported yet but that was years ago so just wondering if there are any workarounds now or if its supported etc?

If it isn't supported, what exactly does the device need to be in order to work, does it need to appear to windows as a USB serial interface? I couldn't find information on it. I tried searching for "Savvycan compatible devices" and nothing was explicit. I ask because I have many microcontrollers and a CAN shield (MCP2515) which I hope I can use. 

I feel more confident with this USB dongle I bought but if anyone can suggest how I could get it working with Savvycan that would be great.

rusoku, 2023-06-10:
Peak , Kvaser, IXXAT, TouCAN .. etc

> If it isn't supported, what exactly does the device need to be in order to work, does it need to appear to windows as a USB serial interface? I couldn't find information on it. I tried searching for [...]

homewsn, 2023-06-10:
As far as I know, CANable (based on STM32F042C6) can be flashed with different firmware versions:
- the original version (https://github.com/normaldotcom/canable-fw) uses the Lawicel SLCAN protocol. SavvyCAN V213 also supports this protocol on Windows 10.
- version of candleLight_fw (https://github.com/candle-usb/candleLight_fw) uses "gs_usb" protocol. On Linux, this is implemented as SocketCAN. As for the Windows version of SavvyCAN, I am currently finishing up the QtSerialBus plugin. I'll post the source code as soon as I'm done, but for now you can try it if you want. To do this, the dll file in the attached to this comment zip archive should be placed in the canbus folder of the [SavvyCAN-Windows_x64_CIBuild.zip V213](https://github.com/collin80/SavvyCAN/releases/tag/V213) distribution. If plugin works, then the connected USB device can be found in the New Connection dialog, connection type is QT SerialBus devices, SerialBus device type is candleLight_fw, port is gs_usb0.
[qtcandlelightcanbus.zip](https://github.com/collin80/SavvyCAN/files/11712738/qtcandlelightcanbus.zip)

nagup, 2023-06-10:
@homewsn Oh yes it actually works. The adapter wants to work, when I set up the connection and close the window. I get about 30 frames captured and then it stops and frames per second drops to 0. I am using an arduino to repeatedly send a message, maybe cause of this its freezing and maybe it wouldn't when connected to an actual vehicle? Although I dont think so but could be a possibility, how have you tested yours?

homewsn, 2023-06-10:
@nagup

I connected two candleLight_fw compatible devices to SavvyCAN, The devices are connected via CAN bus at 500 kbps. Next, in the FrameSender dialog box, I enter something like:
[image]

Then I checked the En box and after a while I got:
[image]

And don't forget about termination resistors in "two-devices home CAN bus".

homewsn, 2023-06-12:
@nagup
The source code and initial release version are available here:
https://github.com/homewsn/candleLight_fw-SavvyCAN-Windows-plugin

nagup, 2023-06-12:
@homewsn That's great, thank you for making this, should hopefully get a chance to use it some time soon