collin80/SavvyCAN #629: Savvycan Appimage not working in Ubuntu 23.04

veee8, 2023-04-28:
I can't seem to get the appimage to open in the latest Ubuntu 23.04. 
I have installed the Fuse Library 2 , and enabled the Execute permissions as usual, but no luck getting it to open with a double click.
I can get it to run from the terminal, but that appears to be pretty buggy, and crashes the whole program when trying to load a DBC file.

collin80, 2023-05-01:
I'll check into it. I'm using Ubuntu 23.04 on my dev machine and it works but I don't compile the appimage, that's done via a CI/CD build chain hosted here on GitHub so maybe something has changed in there. However, this sounds a bit familiar. I've been porting the code to QT6.5 and loading DBC files crashes in QT6 because of a large change in the way some aspects of QT work. The code as it sits in the master branch technically does compile with QT6 now. But, it's broken in many little ways. I checked and the CI build did not use QT6 so that doesn't seem to be the problem here but otherwise I'm not sure why it isn't working. Has your DBC file ever loaded? I wonder if it's a general issue with the actual DBC file you're trying to load (well, an issue with SavvyCAN loading it, perhaps not the file itself).

veee8, 2023-05-02:
I ended up downgrading to 22.04 and the issues went away. I was able to
open the Appimage as intended, and the DBC files loaded just fine and for
the most part everything was working like it should with the SavvyCan
program.
23.04 was more of a hassle than it was worth at this point.

collin80, 2023-06-21:
I have tested the most recent continuous build on Ubuntu 23.04 and it worked for me. Though, I'm using Kubuntu which might make a difference. I remember someone saying that Wayland has difficulties with SavvyCAN. So, I may need to test against a stock version of Ubuntu 23.04 to find the problem.

mulot30th, 2023-10-26:
Here using 22.04  and while I can get saavycan running, I can't get it to open a log file. As soon as I select open log file it instantly crashes every single time

Tried both 1999da8 image and 044fea3  image  both do exact same.

older v2.08  work ok though