collin80/SavvyCAN #639: Error from qmake

nicatec, 2023-05-30:
Hello,
using linux mint and attempting the instructions on the main page, I get the following error. I have hardly any idea what I am doing, however these messages might inidcate actual problems with the source code?

`In file included from connections/canconnection.cpp:3:
connections/canconnection.h: In constructor ‘CANConnection::CANConnection(QString, QString, CANCon::type, int, int, int, int, bool)’:
connections/canconnection.h:319:25: warning: ‘CANConnection::mType’ will be initialized after [-Wreorder]
  319 |     const CANCon::type  mType;
      |                         ^~~~~
connections/canconnection.h:208:9: warning:   ‘int CANConnection::mSerialSpeed’ [-Wreorder]
  208 |     int mSerialSpeed;
      |         ^~~~~~~~~~~~
connections/canconnection.cpp:5:1: warning:   when initialized here [-Wreorder]
    5 | CANConnection::CANConnection(QString pPort,
      | ^~~~~~~~~~~~~
connections/canconnection.cpp: In member function ‘CANCon::status CANConnection::getStatus()’:
connections/canconnection.cpp:261:37: error: ‘class QAtomicInt’ has no member named ‘loadRelaxed’
  261 |     return (CANCon::status) mStatus.loadRelaxed();
      |                                     ^~~~~~~~~~~
connections/canconnection.cpp: In member function ‘void CANConnection::setStatus(CANCon::status)’:
connections/canconnection.cpp:265:13: error: ‘class QAtomicInt’ has no member named ‘storeRelaxed’; did you mean ‘storeRelease’?
  265 |     mStatus.storeRelaxed(pStatus);
      |             ^~~~~~~~~~~~
      |             storeRelease
make: *** [Makefile:2601: canconnection.o] Error 1`

collin80, 2023-05-31:
I believe that the code requires QT 5.15 by this point. They did change some things related to QAtomicInt. 

The code will compile under QT 6.5 if you use the code in the QT6WIP branch. You will likely need to manually select to install the serialbus and serialport QT library modules. They don't get installed by default. This is likely the case for QT 5.15 as well. You should always make sure it is installing serialbus and serial port support as they're not part of the core.

nicatec, 2023-05-31:
Thanks,
I tried all of the online installers I could find under: https://www.qt.io/offline-installers
With Qt6.6.0 and Qt6.5.1 I still get errors ( ‘ParityError’ is not a member of ‘QSerialPort’, ‘FramingError’ is not a member of ‘QSerialPort’, ‘BreakConditionError’ is not a member of ‘QSerialPort’) as well as some warnings.

I could not find an online installer for Qt 5.15.

How can I get the "QT6WIP" branch? Is it open source?

nicatec, 2023-05-31:
Hi,
so I found this version  QT6 development build
https://github.com/collin80/SavvyCAN/releases/tag/continuous-QT6

Which compiles with Qt6.5.1 (current version from Qt installer), no errors.

I dont't actually know where the executable got saved (I am still looking), however I guess the original issue has already been solved.

closed 2023-05-31 by nicatec