collin80/SavvyCAN #649: TXT file not loaded correctly in SavvyCAN

MatinF, 2023-06-30:
Hi Collin,

Attached file contains a number of CAN IDs, but in SavvyCAN the data is not loaded correctly - it appears as if all IDs are `000` or `010`, none of which are in the log file. Can you see what would be causing this issue?
[mercedesw205.txt](https://github.com/collin80/SavvyCAN/files/11914856/mercedesw205.txt)

The file is from a CLX000 data logger.

[image]

collin80, 2023-07-05:
Yes, I will take a look. I haven't had a chance yet but should get to your questions yet this week.

collin80, 2023-07-12:
It loads properly if CLX000 is selected before loading the file. It seems to be autodetect that is failing here. It autodetects as "LAWICEL" which is, of course, 100% wrong. It turns out that the lawicel format is kind of hard to characterize and was overly eager to accept other files that it shouldn't. For the time being, I've moved it to the bottom of the list. All other autodetectors properly know that a CLX000 file isn't for them and reject it. This allows the CLX000 autodetector to recognize it and load the file. Now auto detection works properly for CLX000 files.

closed 2023-07-12 by collin80