collin80/SavvyCAN #654: unable to connect with 83k CAN baudate

rchandrashekar53, 2023-07-17:
Hi,

Does SavvyCAN support 83k (83333) CAN baud rate? I tried connecting it from CANConnectionWindow via "passthrucan" - socketcan but it does not work. However, I do not see issues with other baud rates (125k, 250k and 500k).

Pls share your feedback.

rchandrashekar53, 2023-07-20:
> passthrucan hardoded to 500k
> You need to bulid your own artifact with another speed

I did create configuration for passthrucan with 250k, 125k and 83k.

All the configuration except 83k works fine. Is there any config changes
required apart from can connection?

VenomUA1986, 2024-06-05:
I support the author.
SavvyCan doesn't work on bitrate 83333.
I have two devices.
1. PeakUSB.
2. Lawicel device.
SavvyCan does not work with both devices at bitrate 83333.

At the same time, PCan View and CanHacker work perfectly at this bitrate with the same devices.