collin80/SavvyCAN #661: Crash when try to save logfile

benseller, 2023-09-02:
When clicking the Save logfile menu on Xubuntu the application crashes.

Tried a number of pixbuf problem solutions found on Google, but none helped.

[savvycan_2.13_issue.txt](https://github.com/collin80/SavvyCAN/files/12504114/savvycan_2.13_issue.txt)

collin80, 2023-09-08:
Hmmm, hadn't run into that before. The crash dump makes it appear as if it is a problem getting the GTK resources it is looking for. I know QT tries to do some magic to make file system dialogs like load/save look proper across the various operating systems. I think I may be able to change the behavior a bit to make it draw it's own file dialog instead of trying to make it look native. I'll see if there's any setting I can provide that would help with this.

benseller, 2023-09-10:
> Links:

Collin,

So you know...

Savvy CAN 208 Nov 30 2022 runs fine on the same Xubuntu version.

Ben

collin80, 2023-09-11:
I assume you are using the precompiled versions of each? My best guess is that something changed with the QT version as otherwise not a lot changed that would/should cause this sort of problem. But, the QT version used to compile SavvyCAN is likely to have changed between versions. It tends to keep up fairly closely with the latest released QT5 version. I did find the way to make QT use it's own widgets instead of trying to make native looking file dialogs. I'll see about making this an option and we can see if that solves it or not.