collin80/SavvyCAN #673: Not seeing any can data in main window even tho connection is established

DavidZisky, 2023-09-30:
Hi there! I'm fighting since a few days already trying to get SavvyCan working. I tried on two machines (one MacOS, one windows) with two different CAN devices and even two versions of SavvyCAN (V213 and V208) but I still can't get it to work sometimes. When I go to connection window, I can pick my device, I configure it the same as in other software (that works and shows me incoming can frames), SavvyCan shows connection as "connected" and I can see incoming can messages in "console" window but I don't see any data in the main Savvy window. 
[image]

collin80, 2023-10-02:
Unfortunately, LAWICEL support is pretty crummy in SavvyCAN currently. It almost kind of works for some adapters sometimes if you hold your jaw right and the moon is just right over head. Otherwise it's a mess. So, this is probably not your fault.

DavidZisky, 2023-10-03:
Understood, thanks. I'll try to get hold of some GVRET device then.

closed 2023-10-03 by DavidZisky