collin80/SavvyCAN #675: PeakCAN not setting speed?

collin80, 2023-10-02:
It seems possible that the PeakCAN driver is not taking into account the set speed. 500k seems to work as that's the default but other speeds might be impossible to set. More investigation is in order. But, be warned, Peak adapters might only work at 500k speed right now.

tyalie, 2023-10-18:
Is there any more information on this / why it happens? I'm currently running into this issue.

rusoku, 2023-10-18:
Just tested on the Peak CAN Pro FD adapter.At least 500, 250, 100 kbit works fine for me.

collin80, 2023-10-19:
I still haven't tested it locally. I do have a PeakCAN adapter but I always use it through socketcan on linux so I'm setting it from the command line. I'll have to try this in Windows with the normal PeakCAN driver to see what is going on.

rusoku, 2023-10-19:
Add New Device Connection -> Create New Connection -> Select device (!) -> Change speed to 100000 -> Save Bus Settings.
It works for me.

tyalie, 2023-10-19:
@rusoku did you try it with the "nightly" built?

BitterAndReal, 2023-10-25:
> [1999da8](https://github.com/collin80/SavvyCAN/commit/1999da88c5d6b32bd432b737ba1b60d1a15ff47b)
Do you have a apple silicon (macOS 13.5) compatible build of 1999da8?

rusoku, 2023-10-25:
I have compiled several versions older than this for the arm64 Apple silicon (Qt 5.15.6)
And serial bus plugin for the Rusoku TouCAN USB adapter series only.Rusoku serial bus plugin is compiled like fat file with support x64 and arm64.
I will try this compile (1999da8) version as well in my spare time.

BitterAndReal, 2023-10-25:
Thanks for you help. Will that version support all QT supported CAN dongles?
Is it hard to get the GitHub action build to compile an arm64 Apple silicon version?

rusoku, 2023-10-25:
Qt 5.15.x doesn't have build for the arm64 Qt libraries by default.
Need to rebuild all Qt libraries and plugins before.

rusoku, 2023-10-25:
SavvyCAN for arm64 aka Apple silicon,  commit 1999da8 version:
https://www.rusoku.com/downloads/SavvyCAN_arm64_1999da8.zip
After download need to remove file from quarantine:
xattr -cr FILE_NAME

arm64 version works much faster for me on my M1 AirBook

BitterAndReal, 2023-10-26:
> SavvyCAN for arm64 aka Apple silicon, commit [1999da8](https://github.com/collin80/SavvyCAN/commit/1999da88c5d6b32bd432b737ba1b60d1a15ff47b) version: [...]

Thanks for your help. The app works and I can connect to a CAN bus. 
The weird part is it misses a lot of messages on a busy bus. On a 250kbit bus with 180+ messages per second I only get around 40. If I use SavvytCAN  with parallels on windows I receive all messages.
The app also seems very slow on my M1-Max macBook.

rusoku, 2023-10-26:
I don't see any frames lost.From 500 000 I have received 100% frames with about 780 frames per second on 1Mbit bus speed and 1mS interframe space.If I decrease interframe space down 500uS I get fps speed up to 2000 without frames losses.

 Sender: Kvaser Leaf, receiver: Rusoku TouCAN with QtserialBus plugin and low level USB driver compiled for arm64 Apple silicon.
[image]

BitterAndReal, 2023-10-31:
> I don't see any frames lost.From 500 000 I have received 100% frames with about 780 frames per second on 1Mbit bus speed and 1mS interframe space.If I decrease interframe space down 500uS I get fps [...]

I tried again but my PCAN is not usable with this build. It looses messages.

tyalie, 2023-10-31:
> I tried again but my PCAN is not usable with this build. It looses messages.

Does it loose all messages or just some?

BitterAndReal, 2023-10-31:
> Does it loose all messages or just some? Because I can also report bus instabilities with SavvyCAN + Peak that I don't have when I use the PCAN-View software. Also: Do you use CAN-FD or not, because there's a great difference there in how QT setups the PCan adapter.

I only lose a part of the messages. But quite a big part on a busy bus. The same setup with SavvyCAN and PCAN on Windows works fine.
I have not tried CAN FD.

rusoku, 2023-10-31:
Do you use mac-can dylib for the PCAN ? Which version do you use ?

rusoku, 2023-10-31:
Just tested PCAN and it works too. From 100 000 sent frames (Kvaser Leaf) I received 100% with interframe space 1uS.
Same results as with TouCAN adapter. 
Low level driver: Mac-can libPCBUSB.0.12.1.dylib
PC: Macbook Air M1

anonymouslojz, 2023-12-25:
hi all, we innomaker already update the savvy software to support can fd, you can try and test with it. We including our innomaker can and canfd products. Please check from below github link [https://github.com/INNO-MAKER/usb2can/tree/master/For%20Windows/SavvyCAN-InnoMaker](https://github.com/collin80/SavvyCAN/issues/url)

[image]
[image]

anonymouslojz, 2023-12-27:
> It seems possible that the PeakCAN driver is not taking into account the set speed. 500k seems to work as that's the default but other speeds might be impossible to set. More investigation is in order. But, be warned, Peak adapters might only work at 500k speed right now.

Please test the version we update.
hi all, we innomaker already update the savvy software to support can fd, you can try and test with it. We including our innomaker can and canfd products. Please check from below github link [https://github.com/INNO-MAKER/usb2can/tree/master/For%20Windows/SavvyCAN-InnoMaker](https://github.com/collin80/SavvyCAN/issues/url)

[https://private-user-images.githubusercontent.com/97109297/292757798-83107b8a-f76e-4400-9a74-75df06db7b3f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM2NTIwOTEsIm5iZiI6MTcwMzY1MTc5MSwicGF0aCI6Ii85NzEwOTI5Ny8yOTI3NTc3OTgtODMxMDdiOGEtZjc2ZS00NDAwLTlhNzQtNzVkZjA2ZGI3YjNmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI3VDA0MzYzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc4YjlmMjEwNjRlM2M1YTMzZGQzZGQyOGM1YTkxY2Q2ZDdhZDRmNGM1ZTk3Zjc4Yjg3MTVkODg1OGE3YjZlMzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.M3rg7PB2B5-ghTSxpV65sxcb42r9g-nWm9HpuxikWNY](url)