collin80/SavvyCAN #678: Signal Viewer does not support multiple DBCs?

uhi22, 2023-10-07:
Hi,
in my project I'm using multiple DBC files. They are correctly recognized in the main window, and also the Graphing window perfectly shows the signals of all DBCs.
However, the dropdown list in the Signal Viewer shows only one DBC:
[image]
The first line is the DBC itself, and the second line is the node which is defined in this DBC.

Version: SavvyCAN V208 (Nov 30 2022) on Win10
dbcs: 
https://github.com/uhi22/ccs32clara/blob/main/foccci.dbc
https://github.com/uhi22/ccs32clara/blob/main/foccci_debug.dbc
https://github.com/uhi22/wifican/blob/main/wifican_development.dbc

collin80, 2023-10-10:
I tested this with the most recent CI build and my DBC files show up properly and I can select the nodes, messages, and signals from any DBC file. So, it looks like this was corrected some time between V208 and now.

uhi22, 2023-10-10:
I made a test, added a node into one of the not shown dbcs, and assigned message to the node. Now it shows up. So the point seems to be: If the dbc only contains the defaut node, it does not appear.

clowrey, 2024-10-21:
To fix this problem I had to restart Savvy CAN to have it see the new DBC files.