collin80/SavvyCAN #685: PCAN View log file crash

PhilVoltSport, 2023-10-19:
Hi, I have been trying to load PCAN View files, and then graph signals.
I have a log file with 100,000 lines, which will not import. EVER.
I have trimmed this to 100 lines, which will sometimes import, but sometimes crash. Not so useful through really as it doesn't really show me much....

program just shows the loading dialog box (with a progress bar that does not change) and then quits.

Any insight into this one? running 64bit win10pro, running in administrator mode.
I have tried versions from Nov22, and March23.

Thanks!

collin80, 2023-10-20:
It sounds as if perhaps I'm not loading things quite right according to your file format. I believe I had to code up the PCAN loader merely by looking at a couple of example files, not by having any sort of formal specification. So, it's possible your file has a format that is slightly different from what the program expects. If you can, you might send me the file so I can try it in debugging mode to see why it crashes. 

collink at the domain of kkmfg.com

mitchdetailed, 2023-10-20:
pcan has like 7 different versions of their asc file over the years. it may
be a versioning issue

PhilVoltSport, 2023-10-23:
Hopefully the log file is attached!

[log_file.zip](https://github.com/collin80/SavvyCAN/files/13067513/log_file.zip)

damennix, 2024-11-15:
I have smaller PCAN Viewer files that load just fine. Its just the larger files that cause the crash.

damennix, 2024-11-15:
I found an issue. PCAN Viewer can log more that just Data frames. In my case There were "Event logs" in the file. They overlap with the fixed spacing for Data frames and cause a crash. Removing these lines from the .trc allows the file to load.

  11857    259539.835 DT 18FF0639 Rx 8  00 00 00 00 00 00 00 00 
  11858    259562.535 DT 18FF0939 Rx 8  CB E4 12 00 00 00 00 00 
  11859    279170.610 EV Hardware plug-out
  11860    279773.918 EV Hardware plug-in: PCAN-LAN
  11861    266759.850 DT 18FF0139 Rx 8  C5 17 3C 00 36 00 00 00 
  11862    266760.423 DT 18FF0239 Rx 8  00 00 1C 14 4B 04 00 00

[image]