collin80/SavvyCAN #687: CAN-FD with differing speeds

walnut-engineering, 2023-10-25:
### Discussed in https://github.com/collin80/SavvyCAN/discussions/684

<div type='discussions-op-text'>

<sup>Originally posted by **tyalie** October 18, 2023</sup>
Is it possible to use SavvyCAN CAN-FD with different speeds for CAN and CAN-FD?</div>

There is only a single setting for speed when choosing CAN-FD, but 500k/2M is a common standard now.. in Ubuntu you can set this outside the application but on Windows this is not possible.  Even a config file with the speed configuration would be acceptable.

danjhunter, 2023-11-12:
@walnut-engineering I need to view CAN-FD at 2Mbit/s. Right now on MacOS the max I can set it to is 1Mbit/s. Are you saying you are able to get speeds of 2Mbit/s on Ubuntu? If so how is this possible? @collin80 any idea how difficult it would be to expand beyond 1Mbit/s to higher speeds especially 2Mbit/s?

prajwaltr93, 2023-11-18:
@danjhunter i think you can set it on Linux using IP command, like so 
`sudo ip link set dev can0 up type can bitrate 500000`.

danjhunter, 2023-11-18:
@prajwaltr93 would I not want to do **sudo ip link set dev can1 up type can bitrate 2000000** ? If using Linux can I just do this change to a board I have already set firmware on via macOS? I guess using the ip link function?

danjhunter, 2023-11-27:
I have been able to get this working using CoolTerm @ 1M Baud rate. I am able to set the rate on the board by pressing "h" help and confirming based on the supplied instructions. This displays a settings screen where you can change the parameters. One thing I have noticed is that CANFD frames are only able to display when they are 8 bytes in length. If they are longer I get an error, a bunch of repeating "*" & "!" For message presumably at the 2M speed. I have disabled CAN0 and have CAN 1 set to CANFD at to 500000 and 2000000. I am able to see messages at 500000 but not at 2000000. I have tested this by changing the rate from 2 million down to 500,000. I get the same message stream. I think this probably tied to ESP32RET firmware build 618 not SavvyCAN but figured I would post here as well. If I try to connect to savvyCAN it will connect but due to all of the repeating ! & * "errors" it will disconnect. @collin80 any idea why this might not be working at 2MBit/sec? Is this a firmware issue? Also can you point me to some documentation on the logging functions for the board? Is anyone else using the EVTVcandueV2 for CANFD?

anonymouslojz, 2023-12-25:
@collin80 hi all, we innomaker already update the savvy software to support can fd, you can try and test with it. We including our innomaker can and canfd products. Please check from below github link [https://github.com/INNO-MAKER/usb2can/tree/master/For%20Windows/SavvyCAN-InnoMaker](url)

anonymouslojz, 2023-12-25:
@collin80 hi all, we innomaker already update the savvy software to support can fd, you can try and test with it. We including our innomaker can and canfd products. Please check from below github link [https://github.com/INNO-MAKER/usb2can/tree/master/For%20Windows/SavvyCAN-InnoMaker](url)

[image]
[image]

gedass2000, 2023-12-25:
Do you use the PCAN hardware clone with PEAK libraries ?

anonymouslojz, 2023-12-25:
It Theoretically supports every CAN FD products Qt libraries listed including PCAN,Kavaser.
But what we make is to below products and support other kinds of products by the way (But not very sure if works for those devices.)

***@***.***
 
From: MartinS
Date: 2023-12-25 21:31
To: collin80/SavvyCAN
CC: anonymouslojz; Comment
Subject: Re: [collin80/SavvyCAN] CAN-FD with differing speeds (Issue #687)
Do you use the PCAN hardware clone with PEAK libraries ?