collin80/SavvyCAN #692: Floating type signals not handled properly in DBC editor

Ozuba, 2023-11-03:
Hey i was having a problem related to this issue, while converting the dbc to .c/.h files using cantools the signals defined in the dbc editor as single precision and double precision dont get translated properly to floats in the decoding part.

This may be because the way savvycan handles the floating type extending the endianness section options, i think the proper way shoud be using the ´SIG_VALTYPE_´ section instead.

This past issue talks about it #231

Ozuba, 2025-02-21:
Can some one give a shot to this issue? This brokes heavily the intergration experience with code generators

PDvorak72, 2025-07-29:
I have found, this issue is still present. Cannot set decoding IEE754 float.