collin80/SavvyCAN #712: Mask isn't working on Graphing Window

EMSoe, 2023-12-18:
Hi Collin!

I've tried to use the "mask" function in graphing wondow to clear out the sawtooth signal that is bothering me in the graph that is attached.
[image]
I know that the sawtooth is coming from bits 2 and 3, so I used 0xF3 as it woud the mask b11110011. But nothign changes between using 0xF3, 0xFF or 0x FFFFFFFF in the mask box.

I'm currently using the V213 (built Mar 23 2023).

dontgetfoundout, 2023-12-18:
It looks like the usages of the `mask` parameter have been commented out in the code that creates/appends graphs.

The change was made as a part of 4fd437a44ad7faea94b89cab7d93e0151311b5e5 and b0902c8f50f4d16f2d9af6c09216048f12f05d3f around the same time.  

You could try uncommenting the `//& params.mask;` part at the end of these two lines:

https://github.com/collin80/SavvyCAN/blob/4b46537850b13146359f2cf4fb18feecaedc5592/re/graphingwindow.cpp#L1251

https://github.com/collin80/SavvyCAN/blob/4b46537850b13146359f2cf4fb18feecaedc5592/re/graphingwindow.cpp#L1386

This would theoretically apply the mask but may create other issues.  When a mask value is not provided in the settings window `params.mask` will be 0, resulting in all values being masked out.  Perhaps this is what the commits linked above were trying to address.

A better fix may be to only apply the mask when it's set to a non-zero value.  Something along the lines of:

```c++
        tempVal = Utility::processIntegerSignal(frame.payload(), params.startBit, params.numBits, params.intelFormat, params.isSigned);
        if (params.mask)
        {
            tempVal &= params.mask;
        }
```

dontgetfoundout, 2023-12-18:
I pushed a commit with those changes to my fork of the project for reference:
https://github.com/dontgetfoundout/SavvyCAN/commit/e47b6f216b068c507c35e90a1dfcf6bb9de85def

I'm travelling at the moment and don't have access to hardware to properly validate the changes though.