collin80/SavvyCAN #713: Receiving packets does not work when using Lawicel/SLCAN protocol

rnadyrshin, 2023-12-20:
Hello
I am testing SavvyCAN V208 with a Flipper Zero and MCP2518 based CAN/LIN-module. 
The Flipper application implements the Lawicel protocol, which works perfectly with CanHacker 2.0 and SocketCAN (SLCAN).
In SavvyCAN opening CAN-interface and sending packets works fine, but receiving packets from the CAN does not work. Moreover, in the connection window in the console, incoming packets are visible, but they are not displayed in the main program window.
[image]

Do you have any ideas what the problem might be?
Implementation of Lawicel in the Flipper Zero application: https://github.com/flipperdevices/flipperzero-firmware/tree/d48b107065c087357473a2e4c6cba2bc2bfc18fb/applications/external/can_lin_tools/lib/driver/can_hacker2

esaulenka, 2024-01-19:
Hmm. These messages looks correct. If you able to build SavvyCAN locally (it should be easy enough), try to insert `sendDebug(bla-bla)` somewhere in these lines: https://github.com/collin80/SavvyCAN/blob/da5e5027e3aa14edc52c3fbc5342382856b06b8f/connections/lawicel_serial.cpp#L524-L551

jaytronic211, 2024-02-12:
This was happening when i was using the v213 release. But the latest build seems fine.