collin80/SavvyCAN #718: Importing MF4 files

wakey1991, 2024-01-18:
I have recently bought a CSS CANEdge1 device after previously using CL2000 to log data. I needed the twin channel functionality for my application 

However the output file is MF4. It does not load into savvycan.

I have converted to csv, txt, asc file types and none of them load into savvycan.

Is there a way I can load the file in easily 

Thanks

wakey1991, 2024-01-18:
Update - converting to .asc works on the latest version of savvy can

carlbomsdata, 2024-11-21:
You are a life saver! Big thanks!!

MatinF, 2025-01-07:
Martin Falch from CSS Electronics here. If you record data from a [CANedge](https://www.csselectronics.com/pages/can-bus-hardware-products) in the MF4 format, we recommend using the [MF4 converter](https://www.csselectronics.com/pages/mdf4-converters-mf4-asc-csv) ``mdf2asc`` to convert the [MF4](https://www.csselectronics.com/pages/mf4-mdf4-measurement-data-format) log files into the Vector ASC format. You can then load the ASC files in SavvyCAN for e.g. replay purposes via e.g. a [CLX000](https://www.csselectronics.com/products/can-bus-logger-interface-cl2000) or [CANmod.router](https://www.csselectronics.com/products/4x-can-bus-router-fd-converter-gateway-usb). If you experience any issues with this, please reach out to us via our contact form.