collin80/SavvyCAN #725: SavvyCAN crashes on attempting to load DBC file

rtborg, 2024-01-22:
I'm running SavvyCAN-da5e502-x86_64.AppImage on Ubuntu 23.10. The application crashes if I try to load a DBC file. Log below:

```
void@return:~/Documents/software$ GDK_BACKEND=x11 ./SavvyCAN-da5e502-x86_64.AppImage 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_static_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgvfsdbus.so
qt.canbus.plugins.socketcan: Cannot load library socketcan: (socketcan: cannot open shared object file: No such file or directory)
qt.canbus.plugins.socketcan: Cannot load library libsocketcan, some functionality will not be available.
Cannot load library socketcan: (socketcan: cannot open shared object file: No such file or directory)
qt.canbus: Can not disconnect an unconnected device.
qt.canbus.plugins.socketcan: Function can_set_bitrate() is not available.
qt.canbus.plugins.socketcan: Cannot apply parameter: 4 with value: 500000.
"SocketCanBackend: No such configuration as 30 in SocketCanBackend"
qt.canbus.plugins.socketcan: Cannot apply parameter: 30 with value: 0.
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_static_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgioremote-volume-monitor.so
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_static_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgvfsdbus.so
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_static_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgioremote-volume-monitor.so

(SavvyCAN-da5e502-x86_64.AppImage:93832): Gtk-WARNING **: 11:13:28.833: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Adwaita/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Adwaita/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Aborted (core dumped)

```

N95JPL, 2024-03-13:
Getting the same error here too
`/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_static_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgvfsdbus.so
/usr/lib/x86_64-linux-gnu/gio/modules/libgiognutls.so: undefined symbol: g_source_set_static_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgiognutls.so
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_static_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgioremote-volume-monitor.so
/usr/lib/x86_64-linux-gnu/gio/modules/libgiognomeproxy.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgiognomeproxy.so
/usr/lib/x86_64-linux-gnu/gio/modules/libgiolibproxy.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgiolibproxy.so

(SavvyCAN.AppImage:37836): Gtk-WARNING **: 18:45:10.430: Theme parsing error: gtk.css:1423:23: 'font-feature-settings' is not a valid property name

(SavvyCAN.AppImage:37836): Gtk-WARNING **: 18:45:10.432: Theme parsing error: gtk.css:3309:25: 'font-feature-settings' is not a valid property name

(SavvyCAN.AppImage:37836): Gtk-WARNING **: 18:45:10.433: Theme parsing error: gtk.css:3771:23: 'font-feature-settings' is not a valid property name

(SavvyCAN.AppImage:37836): Gtk-WARNING **: 18:45:13.327: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Yaru/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Aborted (core dumped)
`

N95JPL, 2024-03-18:
Temporary fix to the crashing issue, rebuilt the source code on Ubuntu Mantic and all is well!
Unzip the file and run :dancers: 
[SavvyCAN.zip](https://github.com/collin80/SavvyCAN/files/14640015/SavvyCAN.zip)

SebKuzminsky, 2024-05-13:
Same error here, running CI build of latest master (67dfaf1) on on Ubuntu 22.04:
```
(SavvyCAN-67dfaf1-x86_64.AppImage:3475483): Gtk-WARNING **: 12:18:50.308: Could not load a pixbuf from
 icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NUL
L): Failed to load /usr/share/icons/Yaru/16x16/status/image-missing.png: Unrecognized image file forma
t (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (er
ror == NULL): Failed to load /usr/share/icons/Yaru/16x16/status/image-missing.png: Unrecognized image 
file format (gdk-pixbuf-error-quark, 3)
Aborted (core dumped)
```
This issue affects the AppImage build by GitHub CI, but running a "bare" Linux executable built by hand does not have the issue.

SebKuzminsky, 2024-05-13:
The Windows build of 67dfaf1 is not affected by this issue.

jcoalson, 2024-06-27:
Also crashes for me on Mac for these versions:
V208
V213
dev build 53c0dae (2024/05/13)

However V199.1 does not crash. Unfortunately for me, there's no SLCAN support in that version 😞

phatpaul, 2024-10-01:
Same crash here on xubuntu 24.  (using xfce and x11)  I did not see the issue before upgrade from Ubuntu 22.04 to 24.04

Both release and CI version of V213 crash, but V208 does not crash.

<details><summary>Details</summary>
<p>

 ./SavvyCAN-V213-x86_64.AppImage 
Gtk-Message: 15:36:21.909: Failed to load module "xapp-gtk3-module"
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_static_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgvfsdbus.so

(SavvyCAN-V213-x86_64.AppImage:146186): Gtk-WARNING **: 15:36:22.022: Theme parsing error: gtk.css:1356:88: 'font-feature-settings' is not a valid property name
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_static_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgioremote-volume-monitor.so
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_static_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgvfsdbus.so
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_static_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgioremote-volume-monitor.so

(SavvyCAN-V213-x86_64.AppImage:146186): Gtk-WARNING **: 15:36:26.779: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/elementary-xfce/status/16/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Aborted (core dumped)

</p>
</details>