collin80/SavvyCAN #754: Signal Viewer doesn't seem to support multiplexed signal

Tichael, 2024-03-16:
Using the Signal Viewer, I'm watching the value of a particular multiplexed signal and I can see it toggling between the actual value (seen in the interpreted frame) and a wrong value read when the bits represent something else. I verified that this happen with different frames and signals that are multiplexed, but everything is fine with non-multiplexed signals.

Running SavvyCAN build continuous d5a94a1.

collin80, 2024-03-19:
Yes, that sounds like a pretty obvious bug. I've updated the label and will get it fixed when I can.

Tichael, 2024-03-19:
> Yes, that sounds like a pretty obvious bug. I've updated the label and will get it fixed when I can.

Actually, I decided to fork the repo and fixed it. I will try to fix a couple of other annoyances I found and I will propose a pull request when it's ready!

collin80, 2024-03-19:
I just fixed it too. It's basically a one line fix as the code already had a routine that will return a boolean telling you whether a given signal is really within a frame or not. It's committed as 77f889e4d765d82242f200f4c1187f99a67eff9e and I also just fixed a couple of other little issues. But, I'm sure there are MANY more bugs and annoyances yet to fix.

Tichael, 2024-03-19:
Perfect! The graph window also has the same issue with multiplexed signals (fixed on my side too). I'll merge and see what's left.