collin80/SavvyCAN #757: SavvyCan won't startup on Win10

TurboStew, 2024-03-27:
Hello, I tried to sign up on the Savvycan forum, got no email to confirm, see no "resend" for confirmation email.  Tried to sign up again  and says my email is already taken, so I am in limbo....  I know this is not the best place for the question but see no other option, so here it goes.   I tried it on Win10,  run the exe, it spins for a second, then nothing.  I then got it working using Linux VM, later it started working on win10 and used it several times.  Tried to run it 2 months later and it acts like the first time I tried Windows Savvycan.  It spins for a second, then nothing.  I tried grabbing various builds from old to bleeding edge with nothing.  I had the same issue when I first tried Win ver, went to linux for a while, then tried win and it worked....  2 months later, no work again.  Tried all the usual, disable firewall, run as admin, run from command line, did a deep Google search for answers, nothing....  Any help would be appreciated.

Thanks,
Doyle

TurboStew, 2024-03-27:
I can get the QT6 version to work, but the file playback seems to crash it no matter what I try.

collin80, 2024-03-28:
The QT6 version is somewhat of an experimental test version and does seem to have a lot of issues. But, the QT5 version should have worked so that's strange. Does it give any extra textual output when you run it from the command line? It must be complaining about something but I'm unsure as to what. I know that Windows will tell you that it doesn't recognize the executable and that things aren't signed but it still should run after you tell it to do it anyway. 

I just booted up a virtual machine of Win10 that didn't have SavvyCAN installed and was able to run it after downloading it. So, it seems that it should be functional. Maybe try deleting your preferences file.

It's at <user directory>/AppData/Roaming/EVTV/SavvyCAN.ini

This may help also if you might have accidentally moved the window off screen, perhaps to a second screen you don't usually use. You see, it isn't smart enough to realize this has happened and so if you save your window locations they can get saved to a location impossible for you to see. But, deleting the preferences file will fix that.