collin80/SavvyCAN #758: README.md Update?

FurkanEdizkan, 2024-03-28:
I didn't tried but Qt 6.6.3 is available, should we install Qt5 version or we can use Qt6>

> NOTE: Qt6 currently lacks support for QtSerialBus and many other Qt sub-features. At this time you cannot use Qt6 to compile SavvyCAN. Support for Qt6 should be possible around Qt6.2.

rusoku, 2024-03-28:
At least Qt 6.6.1 QtSerialBus with SavvyCAN Qt6 ver works fine.
I have ported this version for Rusoku TouCAN adpaters (Win and MacOS)
https://www.rusoku.com/products

collin80, 2024-03-28:
It's true, QT6 has long had the necessary plugins to be able to support everything SavvyCAN needs and multiple people have gotten it working. However, the default branch of SavvyCAN still does not actually work right. There is a QT6WIP branch that should successfully compile with QT6 (I think it needs QT 6.4 or greater) but it isn't well tested and may be broken in various places.

FurkanEdizkan, 2024-03-29:
Thank this will help me

closed 2024-03-29 by FurkanEdizkan