collin80/SavvyCAN #759: core dumped

jelasin, 2024-03-28:
I got this err when I try to Load log file
[image]

jelasin, 2024-03-28:
and,  when i run it in my ubuntu:22.04, it's doesn't work too.
[image]

collin80, 2024-03-28:
Unfortunately, for reasons I've yet to dig into, using Wayland messes everything up. So, one way to get a lot closer is to use X11 instead. I actually run Ubuntu 23.10 but I use KDE with X11 and SavvyCAN works fine with that combination. I'm aware that Ubuntu by default uses Gnome and probably defaults to wayland these days too. So, I really do need to get this figured out but haven't yet.