collin80/SavvyCAN #764: Lawicel does not work / BSOD on Windows 11

psykokwak-com, 2024-04-11:
Hi all,
I use an ESP32 with the following library to simulate the lawicel protocol : https://github.com/autowp/arduino-canhacker
It works well with CANHacker on Windows. I can open/receive/send CAN data.
But on SavvyCAN (I tried several versions) it just does not work : 

1/ I receive no data from the device : 
[image]
(Here I should receive the ack of the commands).

2/ When I manually send data (like "v" on the previous example), the command is sent twice!

3/ And the best : When I press the "reset selected device" button. My computer crash with BSOD.

Is someone made Lawicel working on Windows ?

psykokwak-com, 2024-04-11:
Ok, I think I got the point.
This line : https://github.com/collin80/SavvyCAN/blob/0447ec1435e98420f9afaca0a5e55728e02278e0/connections/lawicel_serial.cpp#L239
Is it possible to have another windows build with NoFlowControl instead of HardwareControl ?
Or maybe an option in the configuration menu to choose if we want the hardware flow control ?
I'm not able to build the application myself :(

psykokwak-com, 2024-04-12:
Finally I successfully built it for Windows and it works now.

HerrRiebmann, 2024-04-12:
Good to know :)
I was also investigating why nothing was really sent (and received) to the serial port. Neither by the initialization, nor by manually sending as Text.
Using any terminal application worked fine!

Reusing the connection after restarting SavvyCan also does no work. I allways need to delete and create the connection.

Try to compile it myself as well, to proove this is the solution.

HerrRiebmann, 2024-04-12:
Thank you very much for your compiled version!
But it still does not sent anything to the Arduino:
[image]

And manually sending "Send Text" stopped working. Nothing appears on the Device Console:
[image]
HTerm still empty...

Sending via different Terminal:
[image]

psykokwak-com, 2024-04-14:
Here a new binary that also add a microsecond timestamp to the frames : 
[SavvyCAN.zip](https://github.com/collin80/SavvyCAN/files/14969843/SavvyCAN.zip)

HerrRiebmann, 2024-04-15:
Great! Several restarts and re-buring the Arduino later, it finally responds with your latest version 👍
Thank you for your initiative to just fix it yourself and for me!

If you have anything else I can test, please let me know!

Some things which might be considered in the future (_in general_):

- Extended Messages missing timestamp
- Save/Restore Connection does not work after application restart
- Setting up new Connection offers CAN-FD, but check is not submitted to the Connection settings
- Save Bus Settings does not update anything via Serial
- Reset Connection only disconnects (similar to restored connection)
- Debug Stream sends hundreds of "MERRF" messages (did not find anything about that)

Maybe Collin can have a look into your fix and check those tiny optimizations.