collin80/SavvyCAN #768: SavvyCan is crashing after "Load file" is pressed

galsray, 2024-04-17:
SavvyCan is crashing after "Load file" is pressed. What it can be?
[image]

collin80, 2024-04-18:
It is likely an error in the file loader for the file format you're trying to use. Some of them are pretty brittle when it comes to unexpected file formats. What sort of file were you trying to load?

galsray, 2024-04-19:
Even the file selection window does not appear. I can't add any csv file.

ollybowman, 2024-05-19:
Having the same issue on my mac. Load dialogue box won't even appear. Running the latest version of Mac OS

collin80, 2024-05-21:
Hmm, I thought it was working properly on the current version of MacOS. Can you send the crash dump that MacOS makes? I think it takes a couple of steps to get to it (you have to say you want to send a crash report to Apple then there should be a way to actually get the dump and not send it to Apple). This would then explain what is crashing.

galkinvv, 2024-08-20:
I had the same issue with appimage buildrunnign on Ubuntu 24.04.
Appimage provides some gtk libs as a dependency, thats fine but the problem aries with file open dialog: qt detects that ubuntu is running gnome and tries to open some system file dialogs based on gtk libs. those file dialogs trying to load some gtk plugins (pixbuf icon loaders) from host system and this crashes since they depend on newer versions of **gio + gdk_pixbuf** libraries. But savvycan already loaded the older ones.

Now about workarounds & fixes.
A quick hack that helped me is enforcing preloading system version of gio and gdk_pixbuf:
```
LD_PRELOAD=/lib/x86_64-linux-gnu/libgio-2.0.so.0:/lib/x86_64-linux-gnu/libgdk_pixbuf-2.0.so.0 ./SavvyCAN-6d5bc49-x86_64.AppImage
```
This way the file selection dialog do not crash on openening.

From the developer point of view the fix seems to be more complex but still possible. Loading any system-based file dialogs as a libraries into a appimage is not going to work stably, so I suppose that Qt should be tuned to enforce using non-native but qt provided file dialogs on linux. the only exception can be portal-based file dialogs that are implemented as IPC, not as in-process solution, but it may be to cumbersome and uncompatible especially with older systems. So enforcing Qt-only builtin file dilalogs without loading system plugins would be most stable way for Appimage.